y-yosuke/formula-ppoino #2: EV Module

y-yosuke, 2015-11-04:
ppoino-cars の formula-ppoino にある EV 関連の話のつづき．
https://github.com/y-yosuke/ppoino-cars/issues/4

y-yosuke, 2015-11-04:
formula-ppoino だけではなくて他の作りたいクルマでも利用できるものにしたい．

y-yosuke, 2015-11-11:
【メモ】
SiC半導体インバータの利点
https://www.youtube.com/watch?v=hzS-GOyyhTQ

y-yosuke, 2015-11-11:
【メモ】
Zytek Motors and Inverters / Mugen Honda CR-Z?
http://www.zytekautomotive.co.uk/products/electric-engines/
（あまり仕様的にはフィットしなそうだけど...）

y-yosuke, 2015-11-11:
おっと，YASA Motors に新しい P400 という製品が加わっている．
http://www.yasamotors.com/products/
http://www.yasamotors.com/products/p400_series/

y-yosuke, 2015-11-11:
YASA Motors - Patent Drawings
http://www.google.com/patents/US20130147291

y-yosuke, 2015-11-25:
【メモ】
乗用車やバスなどの商用車のモータ・インバータ・ビークルコントロールユニット
http://tm4.com/products/

y-yosuke, 2015-11-25:
【メモ】
シーメンス Siemens - SIVETEC モータ・インバータ・コントロールユニット
http://w3.siemens.com/topics/global/en/electromobility/Pages/powertrain-ecar.aspx

y-yosuke, 2015-11-25:
Tesla Roadster のバッテリー構造
http://www.teslamotorsclub.com/showthread.php/3810-Roadster-battery-%28ESS%29

y-yosuke, 2015-11-27:
理想的なインバータ・モータ統合モジュール
YASA 750 x2 x2 だったらこんな感じ．
[image]

y-yosuke, 2015-12-21:
Nissan Leaf NISMO RC が載っている MotorFan Illustrated Vol.58［Lite版］が ￥0 になってます．
http://3a.as-books.jp/books/info.php?no=MFI20110716
（紙版より企画を一部抜粋しての提供）

y-yosuke, 2015-12-24:
Formula-e - NEXTEV TCR Formula E 001 (2015-2016) に使われている？？？
http://www.gkn.com/driveline/our-solutions/edrive-systems/documents/datasheets/af-130-tech-sheet-v13.pdf

y-yosuke, 2015-12-24:
Formula-e のバッテリーは

容量 | 28kWh
----|----
出力 | 200kW
----|----
セル重量 | 200kg
----|----
バッテリーモジュール全体重量 | 345kg
とのこと．
「Motor Fan illustrated特別編集 Motorsportのテクノロジー 2015-2016 電子版」より
http://3a.as-books.jp/books/info.php?no=MFS20151224

y-yosuke, 2015-12-24:
Formula-e のバッテリーモジュール結構重いな．
あと F-1 もドライバ込み最低重量 702kg も今はあるのか... 500kg は遠き昔となりにけり．

emijah, 2016-01-08:
安全性に関してうるさくなりましたからね。重量をある程度上げないとドライバを守るのは難しいのかもしれませんね。

y-yosuke, 2016-01-08:
F1のレギュレーションの変遷のまとめを見ると
ざっくりと 620kg ぐらいまでは安全性の確保のため
700kg への変化は駆動系の複雑化のように見受けられます．
http://matome.naver.jp/odai/2134296850068589401

y-yosuke, 2016-01-08:
VW マイクロバスコンセプトカー BUDD-e
  - 蓄電容量101kWh
  - 4輪駆動
  - 最高速150km/h
  - 最大600kmの走行可能

http://response.jp/article/2016/01/06/267337.html

スペックは実走値なのか？ コンセプトとしてはこの値ということなのか？

y-yosuke, 2016-01-08:
Audi e-tron quattro concept

- 4輪適宜駆動
- 3モーター 
  - 合計最大出力 370kW
  - 最大トルク 500Nm
- バッテリー容量 95kWh
- 最長航続距離 500km

http://car.watch.impress.co.jp/docs/event_repo/ces2016/20160108_738120.html

大きめの乗用車だと 100kW 前後が当たり前になってくるのかな．

emijah, 2016-01-13:
Teslaってスペック出ていますよね？ Xとかそれなりのモータが積まれていそう

y-yosuke, 2016-01-13:
https://en.wikipedia.org/wiki/Tesla_Model_X#Specifications

emijah, 2016-01-13:
３８６kWか568kWの２バージョンですね

y-yosuke, 2016-02-19:
日立造船 : 耐久性と安全性を向上させた全固体リチウムイオン二次電池を開発～低温から高温まで動作可能かつ長寿命～
http://www.hitachizosen.co.jp/news/2016/02/002027.html
（寸法と使用可能温度以外の仕様が発表されていない）

y-yosuke, 2016-02-20:
昨年の秋ぐらいに SiCインバータの回路モジュールのサンプル出荷が始まったはずなんだけど...
製品のカタチをなすのにはまだ時間がかかるということなのかな？
（続報はどこかにありましたか？？？）

y-yosuke, 2016-03-15:
【参考】Torque vectoring の話など 
Mercedes SLS Electric Drive. Can Volts Ever Match Pistons? - /CHRIS HARRIS ON CARS 
https://www.youtube.com/watch?v=IElqf-FCMs8

y-yosuke, 2016-03-18:
わーぉ Ariel Atom が電気自動車になっとるー
http://www.wrightspeed.com/about/x1/

y-yosuke, 2016-03-18:
drayson racing の YASA 750 は冷却の配管を削り物の部品で置き換えて高さを抑えている．
http://speedhunters-wp-production.s3.amazonaws.com/wp-content/uploads/2013/12/06200010/Drayson_Racing_Lola_EV-15-1200x800.jpg
http://www.speedhunters.com/2013/12/high-voltage-assault-battery/

y-yosuke, 2016-03-22:
超イオン伝導体を発見し全固体セラミックス電池を開発―高出力・大容量で次世代蓄電デバイスの最有力候補に― 東京工業大学
http://www.titech.ac.jp/news/2016/033800.html

emijah, 2016-03-22:
@y-yosuke 出てくるのに少なくとも５年かかりそうなんですかね？

y-yosuke, 2016-03-22:
「5年」というのはどこかに書いてありましたか？

emijah, 2016-03-23:
この記事はないですが、たいがい５〜１０年が多いので。探してみる必要がありますね。

y-yosuke, 2016-03-23:
YASA 750 2個使いだったら tm4 CO150シリーズ もありだな．
特にコネクタ類が1面だけに出ているのでレイアウト的に非常に助かる． 
http://www.tm4.com/products/power-electronics/co150-traction-inverter/

（Rinehart Mosion Systems の PM100 や PM150 シリーズは端子などが3面から出ていて難儀）

y-yosuke, 2016-03-28:
【参考資料】
SIP（戦略的イノベーション創造プログラム）
次世代パワーエレクトロニクス（どこでもパワエレ機器で豊かな省エネ社会）
研究開発計画
http://www8.cao.go.jp/cstp/gaiyo/sip/keikaku/2_pawaere.pdf

y-yosuke, 2016-03-29:
全固体電池、10年飛び越し（2015.02.19の記事）
http://techon.nikkeibp.co.jp/article/MAG/20150205/402627/?rt=nocnt
Seeo社は "300Wh/kgの電池パックを2017年に出荷予定" / "350〜400Wh/Kgの実現にもメド" とのこと．

emijah, 2016-03-29:
500kmくらい走れないと遠出をする気にはあまりならないっすからね

y-yosuke, 2016-03-31:
【参考】日立造船が全固体リチウムイオン電池を開発、車載用に2020年に製品化
http://monoist.atmarkit.co.jp/mn/articles/1603/04/news052.html

y-yosuke, 2016-04-29:
電池だけでも 200kg ほどになりそうなバッテリーモジュールの設置に関して思案．

1. 単体で成立するバッテリーボックス（ストレスメンバー）に電池を組み込む
2. モノコック後部を長め（+250mm）にしてその中に直接何分割かした電池子モジュールを組み込む
3. モノコック後部を長め（+250mm）にしてさらにバッテリーボックスも単体モジュール化（非ストレスメンバー）

　　　＊下に各方式の長所・短所を列挙

今のところ私が一番良いと考えているのは 1. の

「単体で成立するバッテリーボックス（ストレスメンバー）に電池を組み込む」

が構成としてシンプルで他車種への応用が高く良いと思っていて，
それでレイアウト・設計を進めている．
（他車種への応用性の高さは妄想工学的に吉☆）

単体バッテリーボックス（ストレスメンバー）レイアウトの図
[image]

-----

1. 単体バッテリーボックス（ストレスメンバー）に電池を組み込み
  - 長所
    - EV後部駆動システムとして完全なモジュール化により他車種への応用がシンプルに
    - バッテリー不具合時にバッテリーボックス全体の交換が容易
  - 短所
    - バッテリーボックスにした場合にはそれも人間の乗る方のモノコックと同等の剛性・強度が必要となりコスト増
      - リチウムポリマー電池の事故時の発火の防止
      - ストレスメンバーとして
    - モノコックとリアフレームが直接締結されないことによる剛性低下

2. 長モノコック（+250mm）内に電池を直接設置
  - 長所
    - モノコックとリアフレームが直接締結されることによる剛性向上
    - モノコック構造体が1つで済むのでコスト減？
    - 重量最小？
  - 短所
    - 他車種にてこのシステムを利用しようとした場合に別途バッテリーの設置方法の設計が必要
    - 不具合時にバッテリーを交換するときに各バッテリ子モジュールを外す必要あり
      - 公道走るだけっだったらこれでも十分か？
      - レースに使うことを考えると不十分か？
    - モノコックが長くなることでラジエタパイプの通しどころの自由度が低い

3. 長モノコック（+250mm）内に単体バッテリーボックス（非ストレスメンバー）を設置
  - 長所
    - 高剛性とバッテリーモジュール交換の両立
  - 短所
    - コスト最大？
      - バッテリーボックスの非ストレスメンバー化でその分は増えないが
    - 重量最大？
    - モノコックとバッテリーボックスの二重化でスペース効率が悪い
    - モノコックが長くなることでラジエタパイプの通しどころの自由度が低い
    - 250kg ほどになりそうなバッテリーモジュールをモノコック内に設置するのが大変そう
    - 他車種への応用性
      - 単体自立バッテリーモジュールの場合よりも低く？
      - 長モノコック内にバッテリー直接設置よりも高い？

y-yosuke, 2016-05-09:
【参考】
既存の足回り構造生かすEV用インホイールモーター、NTNが開発
http://newswitch.jp/p/4611

y-yosuke, 2016-05-30:
McLaren Applied Technologies が Formula-E 2016/17 シーズン向けに SiCインバータ を開発中とのこと．
http://serakota.blog.so-net.ne.jp/2016-05-29

y-yosuke, 2016-05-31:
別添 111 電気自動車、電気式ハイブリッド自動車及び燃料電池自動車の衝突後の高電圧からの乗車人員の保護に関する技術基準
http://www.mlit.go.jp/common/000193253.pdf

車両寸法に関する部分を大雑把に記述すると
- 動力系の取り付け位置
  - 車両前端から 420mm 以上
  - 車両後端から 300mm 以上
- 動力系の規定
  - 駆動用蓄電池
  - 電子式コンバータ
  - 駆動用電動機／ハーネス／コネクタ
  - 走行に係る補助装置（ヒータ、デフロスタ又はパワ・ステアリング等）

PR #07 時点の formula-ppoino ではモータ後端が車両後端から 270mm の位置にあるので車両後端をあと 30mm ほど伸ばす必要がある．

y-yosuke, 2016-06-20:
YASA 750 が R がついて新しくなっている．
http://www.yasamotors.com/products/yasa-750/

emijah, 2016-06-21:
性能は同じなんですかね？

y-yosuke, 2016-06-21:
カタログ値の比較

  | YASA 750（旧） | YASA 750 R（新）
---|---|---
Peak Torque @ 450 A | 790 Nm | 790 Nm
Peak Power @ 700 V | 200 kW | 200 kW
Peak Power @ 450 V | 100 kW | 100 kW
Continuous Torque | 400 Nm | 400 Nm
Continuous Power | 75 kW | 70 kW
Maximum Speed | 3250 rpm | 3250 rpm
Peak Efficiency | > 95 % | > 95 %
Total Mass | 33 kg | 37 kg

あとはカタログのグラフに描かれているレンジが違っていて
 - YASA 750 : Max 700 V → Peak Torque 790 Nm
 - YASA 750 R : Max 750 V → Peak Torque 820 Nm ぐらい

YASA 750 R は YASA 750 の機械的な改良が開発の主目的のようです．
> The YASA 750 R has been developed from the proven YASA 750 and features a revised mechanical design for easier installation and integration, including additional mechanical strength and rigidity.

y-yosuke, 2016-06-26:
YASA 750 R でちょっと嫌なのが軸のところを中空構造に出来なさそうなところ．
前のバージョンだと中空にも出来るようでその中に Tripod Joint を仕込めるので
その前提でレイアウトしていたんだけど変わってしまった．

　＜参考写真＞ http://www.ipmd.net/articles/001121.html
　（まぁ，写真では Tripod Joint が小さく見えるので使い物になるのかは疑問だったんだけど...）

ついでに，YASA 750 R の日本語版パンフレットもあるのか．
日本からの問い合わせがそれなりにあるのかな？
http://www.yasamotors.com/wp-content/uploads/2016/02/YASA-750-Product-Sheet-JPN.pdf

y-yosuke, 2016-06-26:
YASA 750 R でちょっと良いのがセンサだか通信だかのコネクタが円筒面に移動しているところ．
前のバージョンだと軸近くの平面側から軸方向に線が出ていて重ねた時どうするんだろう？？？と気になっていた．

y-yosuke, 2016-07-07:
Xtrac に Torque Vectoring で YASA P400 2つを使った減速機がある．
これで良いではないか!!!
http://www.xtrac.com/sectors/automotive-engineering/automotive/product/145

y-yosuke, 2016-07-07:
YASA のサイトにも掲載されていた．
http://www.yasamotors.com/xtrac-gearbox-transmission/

y-yosuke, 2016-07-07:
過去の YASA 400 と Gearbox の組み合わせの話はこのあたり．
https://github.com/y-yosuke/ppoino-cars/issues/4#issuecomment-131498666
https://github.com/y-yosuke/ppoino-cars/issues/4#issuecomment-135669914

emijah, 2016-07-08:
> 過去の YASA 400 と Gearbox の組み合わせの話はこのあたり．
> y-yosuke/ppoino-cars#4 (comment)
> y-yosuke/ppoino-cars#4 (comment)

Xtracのギアボックス、かなり良さげですね

y-yosuke, 2016-07-08:
Xtrac のギアボックス，ピンポイントでジャストなスペック・レイアウトです．
図面だかデータだか手に入らないかなぁ...

y-yosuke, 2016-07-09:
「画像だけを表示」にしたら大きな画像で見られた＠Xtrac
http://www.xtrac.com/custom/image_manager/1227_TV_DUAL_YASA_P400_1.jpg

y-yosuke, 2016-07-09:
加速度，最高車速の計算．

＜仮定＞
- 車重（搭乗者2名含む）: 1000 kg
- タイヤ径 : 0.6 m
- 伝達効率 : 90 %（実際はもっと低いか？）
- モータ YASA P400 x2
  - Peak Torque (450A) : 390 Nm x2 = 780 Nm
  - Max Speed : 7500 rpm
  - URL : http://www.yasamotors.com/products/p400_series/

すると...
- ギア比を 4.2:1 にすると...
  - 加速度がおよそ 1G（ 9.80665 m/s^2 ）に
    - 一定加速度とすると 0 km/h → 100 km/h（ 27.78 m/s ）が 2.83 秒
  - モータの最高回転数から計算した最高車速がおよそ 200 km/s に

計算合ってるかな？
タイヤが加速度に耐えられるかな？

そして Xtrac の YASA P400C 用ギアボックスのギア比は次のように設定可能
- P1227 INTEGRATED LIGHTWEIGHT ELECTRIC VEHICLE (ILEV) TRANSMISSION 
  - Torque Vectoring : 2.5:1 〜 5:1
  - Differential : 2.5 〜 6:1
  - URL : http://www.yasamotors.com/xtrac-gearbox-transmission/

やはり良いではないか．

y-yosuke, 2016-07-09:
まだ寸法も分らんし，写真を見ての適当モデリングだけど今のディフーザー面の内側に収まってるの図．
やはり良いではないか．
[image]

y-yosuke, 2016-07-09:
（編集中）
ただ，800V 450A が出せるインバータが限られてて

- Rinehart - PM250DZ
  - 840V
  - 600A (Peak) / 450A (Continuous)
  - Size : 523 x 391 x 75 mm (15.4L)
  - Weight : 20 kg
  - URL : http://www.rinehartmotion.com/standard.html
  - PDF : http://www.rinehartmotion.com/uploads/5/1/3/0/51309945/pm250datasheet_1a.pdf

少し電流値が下がって

- SEVCON - Gen4 Size10
  - 800V
  - 400A rms (2 min) / 200A rms (Continuous)
  - URL : http://www.sevcon.com/ac-controllers/gen4-size-10.aspx
  - PDF : http://www.sevcon.com/media/78442/gen4_size_10_may_2016.pdf

y-yosuke, 2016-07-09:
電流 450A で電圧を下げて 400V ぐらいのインバータだともう少しある．

- Rinehart
  - PM150DX : 150 kW, 450 A, 400V
    - Site : http://www.rinehartmotion.com/standard.html
    - PDF : http://www.rinehartmotion.com/uploads/5/1/3/0/51309945/pm100-150datasheet_2.pdf
- BRUSA
  - DMC544 : 200 kW, 600 A, 450 V
    - Site : http://www.brusa.biz/en/products/drive/controller-400-v/dmc544.html
  - DMC534 : 150 kW, 450 A, 450 V
    - Site : http://www.brusa.biz/en/products/drive/controller-400-v/dmc534.html

y-yosuke, 2016-07-09:
小さくて高出力・高効率の SiC インバータは来ないんか？？？

y-yosuke, 2016-07-09:
SEVCON には Gen5 というシリーズもあるのか．Gen4 よりもこっちが新しいのか？？？
http://gen5.sevcon.com/
http://gen5.sevcon.com/Gen5%20Size9%20flyer%20Old%20style%20V152.pdf

emijah, 2016-07-11:
これ使えそうですかね？
https://www.simscale.com/

y-yosuke, 2016-07-11:
@emijah SimScale についてはこちらで https://github.com/y-yosuke/ppoino-cars/issues/8

y-yosuke, 2016-09-07:
John Deere を知りながらスルーしてた．トラクター用なのかな？？？
大きいのは他と変わらないけど．
https://www.deere.com/en_US/products/equipment/power_electronics/power_electronics.page

PDF Files
http://www.deere.com/en_US/docs/pdfs/brochures/pd400-single.pdf
http://www.deere.com/en_US/docs/pdfs/brochures/pd400-dual.pdf

y-yosuke, 2016-12-05:
【参考】ミツバ ソーラーカー/EV 製品
http://www.mitsuba.co.jp/products/solarcarandev

y-yosuke, 2016-12-06:
東芝の SiCB のラインナップが増えている．
http://www.scib.jp/product/cell.htm

[207 more replies]