y-yosuke/formula-ppoino #3: 設計 TODO

y-yosuke, 2015-12-17:
- [] サスペンションジオメトリの検討・決定
- [] モノコック後端位置と電池モジュールケーシングの再検討
- [] Rr フレームの設計
- [] Fr フェンダーの面をきれいに

y-yosuke, 2016-02-12:
ヘッドライトはこれで良い気がしてきた...
[image]

y-yosuke, 2016-02-12:
（空力...）

emijah, 2016-02-12:
Rigid Industriesの棒みたいなLEDをウィンドスクリーンっぽいところの下に埋め込めない？
http://www.rigidindustries.com/led-lighting/91061h
とか

emijah, 2016-02-12:
ミラーに埋め込むのは無理ですかね？

y-yosuke, 2016-02-12:
埋め込むなら Fr Wing 〜 Fr Fender にかけてのところに縦目にしたいです．（従来案）

ミラー自体が無くなる可能性もなきにしもあらずですし，そこがヘッドライトとして光るのはデザイン上のバランスとしてちょっと嫌です．

丸目2つは「眼」としてのアイコン的に良いアクセントになる気がしたので試してみました．

感じ的にはルパンが乗っていたようなクラシックカーとか，
http://caxury.com/613
BMW 600 の Fr Bumper と Fr Light の組み合わせ的な感じとかの乗り．
https://ja.wikipedia.org/wiki/BMW%E3%83%BB600

y-yosuke, 2016-02-12:
ノーズコーン + 丸目2つ は Zagato Zanzara 的な．
http://www.zagato-cars.com/contents/en-us/d623_1969_Fiat_Zanzara_500_Zagato.html

y-yosuke, 2016-02-12:
（サーキットに行ったらフロントのナンバープレートとヘッドライトを外して下さい...）

emijah, 2016-02-12:
このZagatoのミラーの設置場所がすごいっすね

y-yosuke, 2016-02-12:
Zanzara いろいろと攻めてて，格好良くて，可愛くて，ベースが Fiat 500 で大好きです．

emijah, 2016-02-12:
500がベース！
500ってシャシーはフレームでしたっけ？

emijah, 2016-02-12:
Ghia Jollyができるということは下の車台が相当強いんですね

y-yosuke, 2016-02-12:
Fiat 500 はモノコック構造です．
https://ja.wikipedia.org/wiki/%E3%83%95%E3%82%A3%E3%82%A2%E3%83%83%E3%83%88%E3%83%BB500

Zanzara については NAVI 2009年10月号に Zagato の元チーフ・スタイリスト: エルコーレ・スパーダさんのインタビューが載っていて一部を引用しますと...

> 「...チンクエチェントの特徴的なインナーフェンダー（泥よけ）をそのまま外板に流用しつつ、そこにごくシンプルなボティワークを架装。同時にロッドの取りまわしをモディファイし、ステアリングの位置を下げ、本来、500の後席フットスペースに相当する箇所にドライバーのシートを置き、着座位置としました。...」

emijah, 2016-02-12:
そうなのか！
あそこまで切り落としてよくシャシーが持つなぁ。
しかし泥除けの形が今ではありえなくらい、いいですね。

y-yosuke, 2016-02-12:
バイクの 4.5インチ ぐらいのランプサイズに合わせて少し拡大（100mm → 120mm）して位置を調整したの図（ぱっちり眼）
[image]

y-yosuke, 2016-02-12:
フロントに丸要素を置くとするとリアにも丸要素が欲しくなる...
ブレーキランプを丸にするか...（既製品が使える）

y-yosuke, 2016-03-02:
Pull Request https://github.com/y-yosuke/formula-ppoino/pull/4 で出したデータのキャプチャ（一部 Hide）2枚

（やっぱ全体のデータをごっそり更新するのは違和感あるなー）

[image]
[image]

y-yosuke, 2016-03-02:
ヘッドライトとブレーキランプは 4インチ径 の汎用部品が使えないかとレイアウトしてみました．
方向指示器やポジショニングランプなどは未だデザインとかレイアウトとか出来ていないです．

フロントサスペンションはアッカーマン，逆アッカーマンのどちらでも行けそうな風に変更したのと，
ウィッシュボーンのモノコック側をモノコック内バルクヘッドで支えられるようにX方向を揃えたのが主な変更．

y-yosuke, 2016-03-11:
【メモ】
フロントのスプリング・ダンパー類をモノコックの内側に設置するとレイアウト的には色々と優位な天はあるものの，事故になった時にそれらが搭乗者を傷つける危険性があるのでモノコックの外側にそれらは設置する．
（何度も自分の中で同じ結論に至っているようなのでメモ）

y-yosuke, 2016-03-31:
2016年3月末時点でのデータを コミット，プッシュ，プルリクエスト，マージしました．
https://github.com/y-yosuke/formula-ppoino/pull/5

y-yosuke, 2016-04-16:
【メモ】
形や大きさから F3 を設計しているような気になるけど
EV の場合はバッテリーモジュールを 300kg ぐらいで考えていて
そうすると搭乗者も含むと車体全体で 1000kg ほどになりそうなので
どちらかと言うと GT に近い感覚で設計した方が良い気がする．

y-yosuke, 2016-05-01:
4月作業分のコミット・プルリクエスト・マージを行いました．
https://github.com/y-yosuke/formula-ppoino/pull/6

y-yosuke, 2016-05-04:
Wing Flap Adjuster
http://www.primusracingparts.com/wing-flap-adjuster_p_292.html

y-yosuke, 2016-05-13:
リアのコンビネーションランプはこの組み合わせで良い気がする．

IPF - LED TAIL & STOP TL-01
LED 丸形テール＆ストップランプ TL-01
https://www.ipf.co.jp/products/tl01.html

IPF - LED BLINKER UNIT - FRONT : TL-02WUF / REAR : TL-02WUR
丸形LED ウィンカー　フロント:TL-02WUF / リア：TL-02WUR
https://www.ipf.co.jp/products/tl01WU.html

y-yosuke, 2016-05-14:
リアコンビランプはこんな感じ．
IPF - LED TAIL & STOP, Blinker Unit

ちょっとランプにかかるスリーブがあっても良いかも．

[image]

y-yosuke, 2016-05-29:
5月分のコミット・プルリクエスト・マージだん．
https://github.com/y-yosuke/formula-ppoino/pull/7

y-yosuke, 2016-05-29:
全体のサイドビュー
[image]

y-yosuke, 2016-06-12:
やはりリアのナンバープレートは車体フレームにつけるもんなんかー
http://minkara.carview.co.jp/userid/1769468/car/1309784/2583849/4/note.aspx#title

封印もあるし，むやみに外したり出来ないようにしているんだろうけど...
そーなると formula-ppoino の場合は登録的にはリアサブフレームが車両本体のようになるのかな？

y-yosuke, 2016-06-14:
HONDA VFR800P のウィンカーが大きいので正面視で 20cm^2 を越えていたら使えそうなのだが...
とっくに生産終了か... 
https://ja.wikipedia.org/wiki/%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB:Japanese_HONDA_VFR800P_police_motorcycle.jpg

emijah, 2016-06-14:
これはどうですかね？
https://www.amazon.co.jp/gp/aw/d/B0174D9LR8/ref=mp_s_a_1_25?qid=1465895920&amp;sr=8-25&amp;pi=SL75_QL70&amp;keywords=%E3%82%A6%E3%82%A3%E3%83%B3%E3%82%AB%E3%83%BC+%E3%83%90%E3%82%A4%E3%82%AF+%E3%82%AB%E3%82%A6%E3%83%AB

y-yosuke, 2016-06-14:
おそらく 20cm^2 以下ですね．
バイクの汎用品でも探しているのですが大体は車の規格に満たない大きさで不採用になります．

y-yosuke, 2016-06-14:
【メモ】20cm^2 を満たす寸法とは
- 7cm x 3cm = 21cm^2
- 5cm x 4cm = 20cm^2
- ( 5.5cm / 2 )^2  x 3.14 = 23.7cm^2

方向指示器の法規（大きさなど）
http://www.mlit.go.jp/common/000187504.pdf

y-yosuke, 2016-06-14:
Ninja1000 デイライト内蔵LEDウインカー
http://www.bosscom.jp/product/ninja1000/
大きそうだけど寸法が書いていない．そしてべらぼーに高い (T_T)

y-yosuke, 2016-06-14:
ドライブレコーダーはバイク用のが良さげ．例えばこんなの ↓
https://www.yupiteru.co.jp/products/bike_drive_recorder/bdr-2wifi/spec.html

そして補助制動灯の設置（高さ850mm以上）とカメラマウントを兼ね備えた左右のロールバーを繋ぐ位置の整流ウィング風ステイを設計中．
[image]
[image]

emijah, 2016-06-14:
これいいっすね！
ZZRのカウル用のウィンカーって確かでかかったような

y-yosuke, 2016-06-14:
ZZRのウィンカーも大きくてもしかしたらOKなのかも？と思ったのですが
ウィンカーの部品をWEB見たときに売切れだったので
カワサキのサイトを見たら現行車種になかったので除外していました．
http://www.kawasaki-motors.com/mc/lineup/

現行車種でなくても部品の在庫さえあれば検討対象になるのですが
バイクの部品というのは廃番後も長く在庫しているものなのでしょうか？

y-yosuke, 2016-06-14:
もしくはフレキシブルLEDチューブライト的な
http://ja.aliexpress.com/store/product/2pcs-60cm-DRL-Flexible-LED-Tube-Strip-Style-Daytime-Running-Lights-Tear-Strip-Car-Headlight-Turn/121101_32362504399.html?storeId=121101

emijah, 2016-06-14:
こっちの方がいいかも

y-yosuke, 2016-06-21:
ちょっと上がってた集中力が落ちたので6月分としては早めだけど Commit, Push, PR and Merge．
変更の詳細と全体のスクリーンショットを Pull Request 8 に載せておきました．
https://github.com/y-yosuke/formula-ppoino/pull/8

y-yosuke, 2016-06-27:
スチール（SCM？）版 Roll Hoop
[image]

y-yosuke, 2016-07-07:
Bulge - Damper / Fr Suspension
Environment Map（Space Needle）
[image]

y-yosuke, 2016-07-21:
モノコックの足下部分の面を水平にそして少し上げようと考えたが
そうすると車両法規のシートの寸法 40cm x 40cm をクリアするのが難しくなりそうだ．

水平面に投影したときに 400mm x 400mm 四角となるような座面（仮）
[image]

座面とモノコック，ドライバとの隙間の関係．下の条件を考えると Fillet 部分が無いと厳しい．
- モノコック厚さ 25 〜 30mm
- シートのクッション厚さ ?0mm ？
[image]

y-yosuke, 2016-08-20:
さて CFD が SimScale で出来たのでフロント周りのエアロデバイスばかりこのところ試していた #12 が目処も着きそうなのでリアもと思うところだが Xtrac のギアボックスを採用したいのでレイアウトとのものから変更で，フレームはもちろん，サスペンションジオメトリも変わるかもの状態．

Xtrac のギアボックスの CAD データがあれば最も良いが，せめて外径寸法の載った図面がないかな．
でないとレイアウトが進まないぞ．

emijah, 2016-08-22:
聞いてみましょうか？

y-yosuke, 2016-08-22:
Xtrac にですか？ ぜひお願いします．

emijah, 2016-08-22:
了解です。

y-yosuke, 2016-08-22:
ちなみに画像から想像モデリング https://github.com/y-yosuke/formula-ppoino/issues/2#issuecomment-231523173 したのを横から見た図．（ギアボックス→紫線）
コンパクトであって欲しい願望がモデリングには入っていそうなので実際はもう少し大きいようなきがする．
[image]

y-yosuke, 2016-08-23:
#15 で最新のデータをマージしました．

y-yosuke, 2016-08-31:
【メモ】
Technical Regulation などの PDF にある点群の座標データ（テキスト）から Rhinoceros に点群をインポートする手順

1. テキストデータが含まれる PDF ファイルを保存
2. PDF ファイルを Adobe Reader で開いて テキスト形式（*.txt）で保存
3. 必要な座標が描いてある部分のみのテキストファイルを作成
4. LibreOffice でテキストファイルを開く（適宜セルのセパレート書式を指定）
5. LibreOffice にて CSV 形式（*.csv）で保存
6. Rhinoceros で CSV ファイルをインポート

y-yosuke, 2016-09-01:
F3 Wing Profile（R3付）の "中" と "小" を用いた 2pcs Rr Wing こんな感じ．
（ End Plate 上端 1pc 比 +50mm ）
[image]
[image]

y-yosuke, 2016-09-01:
少し引いた図
[image]
[image]

y-yosuke, 2016-11-02:
Ride Height とかまーそーなるわな的に参考になる．
Wheelbase とか Weight distribution とかも．

http://bac-mono.com/specs.php

> Vehicle Specification
> Dimensions and Weight
> Overall width 	1836mm
> Overall length 	3952mm
> Height 	1110mm
> Wheelbase 	2565mm
> Ground clearance 	100 mm (f) 110 mm (r)
> Weight 	580kg*
> Power to weight 	525 bhp per tonne
> Weight distribution 	48% (f) / 52% (r)
> 	
> * Dry weight and subject to options

y-yosuke, 2016-11-18:
ヘッドライトは今は 4.5インチ ぐらいでレイアウトしているけど 5-3/4インチ が一般的なのかな？
http://store.shopping.yahoo.co.jp/vulcans/cps0417.html

y-yosuke, 2016-11-22:
バイク用ドライブレコーダ
http://www.driveman.jp/products/bs8.html

関連コメント
https://github.com/y-yosuke/formula-ppoino/issues/3#issuecomment-225832819

y-yosuke, 2016-11-22:
ユピテルのは加速度の記録をプロット出来るので良いかもしれない．
https://www.yupiteru.co.jp/products/bike_drive_recorder/bdr-2wifi/spec.html

y-yosuke, 2016-11-22:
カメラ2台モニター付 バイク・自動車用ドライブレコーダ
http://www.lisway.com/SHOP/HDVR00081.html

y-yosuke, 2016-12-17:
ちまちましたカウルにするから雨どいをどうするかとか考えなくてはいけないのであって
LMP みたいに，で〜んとでかいカウルにしたらえぇ〜やん!?
（お高い万円になりそう...）

y-yosuke, 2016-12-19:
レースでの使用は付随的なものとして，
ボンネット以外は基本的にファスナー類を使わずにボルト締結で良いだろうか...

y-yosuke, 2016-12-20:
やっぱカウルの設計をしてゆくと小間切れになってゆく... orz

emijah, 2016-12-20:
同じ穴をボルトやファスナーで使いまわせるようになっていればいいんじゃないですかね？
売り先がレース中心になるのであれば、レースよりにふって共通にした方が量産は楽っすよね

emijah, 2016-12-20:
シリコンのテープか何かで封をして防水化するか？

[74 more replies]