y-yosuke/formula-ppoino #9: エンジン駆動としたら

y-yosuke, 2016-06-27:
エンジン搭載の場合は使える適切なエンジン・ギアボックすがあるのか？

- 調達ができるか（試作・量産）
- 設計用のデータがあるか？
- 適切なギアボックスがあるか？

emijah, 2016-06-27:
F4から流用はできなさそうですか？（弱すぎ？）

y-yosuke, 2016-06-27:
せめて F3 と思うのですが... あと，エンジンの長さにもよりますがおそらくは横置きになります．
F3 などと比べるとモノコック後端からリア車軸までは短いので，
そして2人乗りの分，幅方向はあるのでなるべく重心を前にしたいので．

でも，PORSCHE 919 Hybrid の 2L V4 エンジンなんか短かったらNA使いで縦置き妄想☆
http://sportscar365.com/lemans/wec/porsche-reveals-first-photos-of-919-hybrid-engine/
http://www.tomorrowstechnician.com/porsches-le-mans-919-hybrid-turbo-four-cylinder-engine/

emijah, 2016-06-27:
昔のマッスルカーのようにギアボックスが後端から突き出ているというのは無しですかね？美的感覚からアウトかな

y-yosuke, 2016-06-27:
重心はなるべく前で

y-yosuke, 2016-06-27:
バイク用で V4 エンジンがあるか．（ギアボックスつながるのか？）

http://bike-lineage.jpn.org/yamaha/vmax/vmax.html
本当に 200馬力 出るのだろうか？

http://thekneeslider.com/is-the-motus-v4-baby-block-crate-engine-the-future-of-old-school-hot-rodding/
出力値を見つけられなかった．

y-yosuke, 2016-06-27:
まぁ，ドイツでは走れなくなるかもしれませんが...
"ドイツ政府、2030年までに全ての新車を無排気車にすることも検討中"
http://jp.autoblog.com/2016/06/25/germany-cars-emissions-free-2030/

emijah, 2016-06-27:
この会社のエンジンはどうですか？

y-yosuke, 2016-06-27:
リンクなし？？？

emijah, 2016-06-27:
失礼
http://liquidpiston.com/

y-yosuke, 2016-06-27:
Liquid Piston のはまだ大出力化に至ってないと私は認識していますが，どうなんでしょう？

ロータリーエンジンの1つだと思いますが，
一般論としてロータリーエンジンは同出力でも体積が小さくできるメリットがあります．

もしかしたら EV のレンジエクステンダーのように使うのもありかもしれません．

emijah, 2016-06-28:
http://liquidpiston.com/technology/x-engines-diesel/
を見ると40, 70HPは達せしていますが、確かにそれ以上の出力のものは見ませんね。

y-yosuke, 2016-09-16:
ひとまずクローズ

closed 2016-09-16 by y-yosuke