y-yosuke/formula-ppoino #11: CFD Simulation

y-yosuke, 2016-07-11:


y-yosuke, 2016-07-12:
W(Half) : 3.0 [m] x H : 6.0 [m] x L : 30.0 [m] ( Blockage : 3.6 [%] )  だとこんな感じ．
[image]

y-yosuke, 2016-07-12:
SimScale にCADデータを上げた．これからメッシュ切りの手順を勉強する．
https://www.simscale.com/projects/yosukegb4/fp-023b/

y-yosuke, 2016-07-12:
あー External Flow にしたら Box は SimScale 側で設定できるのか．
とりあえずこのデータで Intrenal Flow として今メッシュ切り中．

y-yosuke, 2016-07-12:
データは STP で出力したけど Internal Flow としてメッシュ切るのなら STL の方が良いとかあるのかな？それも後々試そう．

y-yosuke, 2016-07-12:
automatic で fitness を very fine にしても粗い．フラップの隙間が潰れてる．
[image]

emijah, 2016-07-13:
うーん

y-yosuke, 2016-07-13:
おそらく internal flow が原因だと今のところ考えていて external flow にしたらモデル周辺は細かく切って遠いところは粗く切るのではないかと想像しています．
今日それを試してみます．

y-yosuke, 2016-07-13:
あ，YASA 750 のバルジのモデルを入れてない．（ま，メッシュの練習だし...
あと，地面のメッシュは細かく切りたいな．

y-yosuke, 2016-07-13:
メッシュの切り方，丁寧に書いてあるわ
https://www.simscale.com/docs/content/tutorials/tutorial_frontwing.html

y-yosuke, 2016-07-13:
ところで SimScale の Core hours 3000 hours って月あたりの時間ということなのか？否か？
4 cores でフルで回したら 4 [core] x 24 [h] x 30 [day] = 2880 [h] で大体それぐらいの時間になるのだけど．

y-yosuke, 2016-07-13:
メッシュ粗さの件は internal / external ではなくて automatic / parametric の違いみたいだ．
parametric にして領域・面などによってメッシュの疎密を設定する．

y-yosuke, 2016-07-13:
良い感じになりもうした．
もう少し下流の粗いところを密にした方が良いような気もするが，まずこれでCFDにかけてみよう．
[image]
[image]
[image]
[image]
[image]

y-yosuke, 2016-07-13:
で，ノード数が 26109120 になったのだけどシミュレーション計算にかけたら
8 cores ではメモリが足りませんでした．となったので 32 cores で設定し直して再計算中．
それでも駄目だったらメッシュの削減をしなくてはならない．

y-yosuke, 2016-07-14:
32 cores にしてもメモリが足りないとのことでメッシュを粗く切っているところ．

y-yosuke, 2016-07-14:
【メモ】CADデータの受け渡し
- STEP
  - タイヤのような1つの面で大きく曲がっているものの形状の再現性が怪しい．
  - 単位もデータ内に持っているようで [mm] で出力しても [m] に自動的に換算されている．
  - インポートされた面はそれぞれ持って行かれるのでそのまま face set などにできる．
- STL
  - [mm] で出力したらそのままなので Geometry Operation で 0.001 乗算変換
  - インポートされた面は全体で1つの閉曲面として認識されるので Geometry Operation で切り分け角度指定での面分割が必要

emijah, 2016-07-14:
;_;)

y-yosuke, 2016-07-14:
スケール変換なども SimScale 側の core 占有が勿体ないと思ったら MeshLab で予め変換しておけば良い．

y-yosuke, 2016-07-14:
SimScale 上での STL の Surface splitting は Error が出て失敗したので STL は不採用．
よって，STEP の変になる面，具体的にはタイアの Surface を Knot に沿って分割して
SimScale に読み込んだときの質を確保．

y-yosuke, 2016-07-14:
IGES も試したがタイヤのサーフェスがおかしくなるのは STEP と同じで，
更には単位が着いてこないので 0.001 倍しないとならなく，
STEP に比べて良いところがないので不採用．

y-yosuke, 2016-07-15:
【作業メモ】
Fr Flap も SimScale 上の STEP での形状再現性が低かったので Rhinoceros にて Shrink Trimmed Surface を行って STEP エクスポートのし直し．

y-yosuke, 2016-07-15:
1つずつメッシュが正常に切れない問題をつぶしているところ．
- メッシュは曲率の大きさによって精度を分けることにより形状追従性と省メッシュを両立
  - ウィングの前後端は面を分けておく（Rhinoceros だったら Split）
  - いわゆるフィレットの部分のメッシュを細かく
- 面の交差が 90 deg を超えるようなところ？は面の続きがおかしくなり易いので注意
- Wind Screen のような薄いところは SimScale の Geometries -> [ CAD Data ] -> Topological Entity Sets で face set を作っておき Mesh Refinements で特に細かく設定
  - もしくは CAD File の時点で厚くしておく

y-yosuke, 2016-07-16:
もう12年ぐらい前の話だけど当時 SCRYU/Tetra でメッシュ切るのそんなに苦労した記憶がないのに何で？とぐぐったら SCRYU/Tetra の売りはメッシュ切りと当該ページには書いてある．そういうことなの？？？
http://www.cradle.co.jp/products/scryutetra.html

テトラメッシュでも切ってみるか... 
（メッシュ切りのテストだけで 3000 core hours を使ってしまうのでは!? という恐怖... orz）

y-yosuke, 2016-07-16:
OpenFOAM の方から調べると blockMesh は薄い部分の表現が苦手なようだ．

y-yosuke, 2016-07-17:
external flow で Bounding Box と元の CAD Data の組み合わせで流体領域を形成するのだけれども，
どこかのメッシュ（フラップとフェンダーの交差部？かタイヤと地面の交差部？）が裏返ってしまって
領域が車体の内外両方になってしまう．

このため internal flow の CAD Data を再び使い，今回は "Hex-dominant automatic for internal flow (only CFD)" ではなく "Hex-dominant parametric (only CFD)" で良い感じにメッシュが切れないか試行中．

y-yosuke, 2016-07-17:
"Tetrahedral with layer refinements" でメッシュ切りを試したがエラーが出て切れなかった．
理由はわからないが，今回はこの方向の試行は中止かな．snappyHexMesh の方向でがんばる．

y-yosuke, 2016-07-17:
流体領域を CAD Data で形成しておいた方がおかしなことになりにくいようだ．

SimScale 上でライドハイト設定を変えてメッシュが生成されるのがベストだろうが，まともなメッシュが切れないようでは元も子もないので，手間は増えるが予め CAD で流域を形成したデータを複数使うのがセカンドベストかもしれない．

手間と言っても車体モデルを Transfer して流域の Box と Boolean をとるだけだけど．
更には厳密に言うとライドハイトによってサスペンションアームの形が変わるのでそう言った意味ではこれがベストなのかもしれない．
SimScale の CAD データは "Import from Onshape" の機能があるのでライドハイト変化などのパラメトリック要素は Onshape 上で処理をして流体領域を形成して一連のパターンを SimScale にてバッチ解析する作業の流れもありかも．

y-yosuke, 2016-07-17:
メッシュ良さげか!? これで解析に流せるかな．
https://www.simscale.com/workbench/?pid=4105619712994739995#tab_0-0
Meshes -> Mesh 1 2 2

Nodes (Points) 6,245,308 で大体目標値．

メッシュ全体（手前の面は Hide してる）
流域 Box の周辺が細かいのは勿体ないが CAD のエッジをメッシュに反映する設定 "Feature refinement" が CAD の Surface 全部にかける方法しかないため仕方ないか，というところ．
改めて見ると，車体後流部はもう少し長く，細かくしたいところだが...
[image]

メッシュ拡大
Flap と Wing の隙間が怪しいことになってる．改善したいところだが...
[image]

レイヤーの様子（3層）
[image]

領域はちゃんとしているようだ．
[image]

y-yosuke, 2016-07-17:
I think snappyHexMesh is not beautiful.

y-yosuke, 2016-07-17:
まだメモリーエラーが出るなぁ... (T_T)

y-yosuke, 2016-07-17:
やはりメッシュがおかしいような気がする．

SimScale の STEP/IGES → サーフェスメッシュの変換がおかしいのでいろいろな人の Project をみていたら，これ ↓ なんか STL で読み込んでいるのに複数ソリッドが存在している．
https://www.simscale.com/projects/pfernandez/perrinn_f1_v5/
ソリッド化した STL だとそれぞれの閉領域が1つのまとまりとして SimScale で認識されるらしい．

STL でもソリッド化というプロセスがあるらしく，それは FreeCAD で出来るらしい．
http://monoist.atmarkit.co.jp/mn/articles/1602/12/news011.html

ということで FreeCAD 氏に働いてもらっているところだが... 帰ってこない... orz

y-yosuke, 2016-07-17:
複数要素の STL データを SimScale にインポートするために試したことで駄目だったことリスト
- Rhinoceros から Binary でSTL出力する → 1つのオブジェクトになるのは変わらず
- 複数の STL ファイルを ZIP にまとめてアップロードする → 最初の1つの STL ファイルしか読み込まず

emijah, 2016-07-18:
うーん、互換性にまだ問題があるんですね

y-yosuke, 2016-07-18:
タイヤ＆ホイールを回さなければ STL でそのまま出来るようなのですが...

y-yosuke, 2016-07-18:
3000 core hours は1年あたりとのこと．すぐ使い切りそうだな．
"3000 core hours of computing power per year"
https://www.simscale.com/forum/t/simscale-plans-pricing/27827

y-yosuke, 2016-07-18:
STL で複数の領域があるデータを SimScale にアップロードする方法が分った．

SimScale のフォーラムに書いてあったのだがそれだけでは足りない部分が私の環境ではあったので Rhinoceros で STL 出力してから SimScale にアップロードするまでの手順を書いておく．

### 1 . STLファイルのエクスポート（ASCII）

Rhinoceros で分割して利用したいソリッド部分をそれぞれ **ASCII で STL 出力**する．
例えば，body, fl-wheel, fr-wheel, rl-wheel, rr-wheel などで分ける．

### 2 . STLファイル内の solid 名の変更

STLファイル内の solid の名称が全て OBJECT になっており，そのままだと SimScale で利用する際に区別がつかなく face の選択ができない．
そこで，各 STLファイル をテキストエディタなどで編集してファイルの冒頭と末尾にある次の行を編集して，各 solid が固有の名称を持つようにする．

＜変更前＞
```
solid OBJECT
...
endsolid OBJECT
```

＜変更後（例）＞
```
solid OBJECT2
...
endsolid OBJECT2
```

### 3 . STLファイルの結合
コマンドラインで次のように cat を実行して複数のSTLファイルを1つのSTLファイルに結合する．
```
cat 1.stl 2.stl 3.stl 4.stl 5.stl > fp-023b_cfd_combiled_rename-OBJECT.stl
```

### 4 . ZIP圧縮（必要ならば）
結合したSTLファイルが 300MB を超えるような場合はZIP圧縮をして 300MB 以下にする．

### 5 . SimScale にアップロード
SimScale の "Upload CAD/mesh" の "Upload geometry" から STL を指定して結合したSTLファイルをアップロードする．
アップロードしたデータはSTLファイル上では solid と記述されているものは face として SimScale 側に認識されるようでそれを使って各領域のメッシュ条件や，解析の境界条件として利用する．

アップロードしたデータの様子
[image]

さて，やっとここからメッシュ切りのやり直し．

y-yosuke, 2016-07-18:
なんや... やれば出来る子やないか...

（でも，細かくしすぎたので再メッシュ）
[image]
[image]
[image]
[image]

y-yosuke, 2016-07-18:
メモリーエラーはメモリーエラーにあらず，の場合もあるとのころ．
https://www.simscale.com/forum/t/memory-error-whilst-trying-to-simulation/28038/6

y-yosuke, 2016-07-18:
今度は粗くしすぎたかなーという感はあるが，とりあえず解析が出来るようにするのが第1目標．
[image]
[image]
[image]
[image]

y-yosuke, 2016-07-18:
とりあえず解析で解は収束方向に行った．
Cd = 0.594122645998
Cl = 0.139484758275
2000回の計算を 32 cores で 178min（約3時間 = core hours 3.2% 使用）

デバッグのためタイヤ＆ホイールを回していなかったので回して再度解析する．

- 結果のファイルが 1GB ぐらいあるようなのでダウンロード，保管も厄介
- Web の Post-Processor は非常に重いので（そらそうだ）ParaView などで結果を見る

タイヤ＆ホイールを回していないケースだけど解の収束の様子はこんな感じ．
[image]

y-yosuke, 2016-07-18:
core hours が 3000 h/year を使い切りそうだったら複数人で free アカウント持ちよりで手分けして解析かな．SimScale 上でブランチ出来るみたいだし．協力者いるかしら？

y-yosuke, 2016-07-19:
寝る前に仕込んだのが計算は出来ているのに何故エラーで結果がダウンロード出来ない状態だった．
この原因かは分らないが Simulation Run の名前が原因で起こる場合があるとのこと．
名前は結果が出てから rename するのが吉らしい．（そして再計算）

> For some reason the / character in the naming of the Simulation Run 1 m/s was the source of the error. As you can see when the run was renamed Run 2 it worked (so your setup was perfectly fine!)
https://www.simscale.com/forum/t/lego-technic-24-hours-race-car-simulation-project-by-pfernandez/4158/6

y-yosuke, 2016-07-19:
タイヤ＆ホイールを回転させた結果がダウンロード出来なくても Solver Log に Forces and moments の結果は残っているのでコピペ．

計算合っているのだろうか？ Cm から考えるに Cl(f) と Cl(r) が逆なような気もしないでもない．Cm が逆なのか？
```
forces forces00 output:
sum of forces:
pressure : (196.251288684 -198.050836462 -65.3098265119)
viscous : (16.7956659701 -1.11352759496 0.679483712905)
porous : (0 0 0)
sum of moments:
pressure : (75.8217441964 133.719615886 73.5065703079)
viscous : (-0.0170156293263 9.10674333444 10.8821343082)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.0956212772405
Cd = 0.567681285646
Cl = 0.172212910301
Cl(f) = 0.181727732391
Cl(r) = -0.00951482208985
```

y-yosuke, 2016-07-19:
Lift direction を (0,0,-1) では逆か．Downforce direction じゃないしな．
Lift direction (0,0,1) でないと座標系がおかしくなりそうだ．
Center of rotation もこれで合っているのか？？？
[image]

y-yosuke, 2016-07-19:
計算時間の感覚は今のメッシュ 6M nodes ぐらいで 32 cores つかうと
- メッシュ 1時間弱
  - もかからないかな？ Usage Overview を見ると 30 min とある．
- 解析 3時間

y-yosuke, 2016-07-19:
仮に既出の coefficient の計算は合っているとすると前後の空力バランスがアンバランス過ぎるので何か設計上の対策をしなくてはならない．

 Clf と Clr は逆だったとするとリアがほぼ 100% なのでこれを重心に合わせる感じで 60〜70％ まで落としたい．検討する点を忘れないうちに書いておく．

- Rr Bumper のL字形断面形状のダウンフォースの発生度合いを解析結果から検討
  - L字形によるダウンフォースの発生が大きいようならボックスに近い形状への設計変更
- Monocoque の下面がリフトを発生しやすい形状なのでその点を解析結果を検討
  - リフトが大きいようだったら下記デバイスなどの検討
    - Side Deflector の Stepped Bottom Plane 上の Fin
    - Forward Side Deflector
    - Nose Cone S-Duct
    - Bib

Clf と Clr が合っている場合はフロント100%なのであまりあり得ないような気もするがその場合は...
- Fr Flap の小型化
- Rr Wing を高く

emijah, 2016-07-19:
協力しまーす

y-yosuke, 2016-07-19:
ありがとうございまーす．
core hours あと 1500h ぐらい残っているのでもう少し先だと思いますが，
使い切る前にメッシュ切り・解析の道筋を確定させます．

y-yosuke, 2016-07-19:
Cl が小さいので発生モーメントのほとんどはドラッグが原因説
（Clが小さいのは一応意図どおり？）

y-yosuke, 2016-07-19:
これはほとんどがフロントのダウンフォースかもしれない．
リアボディンの圧力の感じからすると地面から遠いし，ウィングは低いし角度ないしオーバーハングないしで本当にリアダウンフォースがないかもしれない．

Lift direction を (0,0,1) に直してもう一度解析してまた考えるか．

[image]
[image]
[image]
[image]

y-yosuke, 2016-07-19:
収束の感じからすると1500回の計算で良いように思う．

y-yosuke, 2016-07-19:
ちなみに解析計算がちゃんと出来たところの SimScale のリンク
https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/

y-yosuke, 2016-07-19:
ParaView 氏はアニメーション動画を描き終わった直後に，力尽きたか自らお閉じになられた．

y-yosuke, 2016-07-19:
Lift direction を (0,0,1) にしたときの力の計算結果．Clf と Clr が (0,0,-1) の時と逆になっているなぁ．
```
forces forces00 output:
sum of forces:
pressure : (196.29148201 -197.586939295 -65.3514275332)
viscous : (16.7948184932 -1.11385066373 0.675038074106)
porous : (0 0 0)
sum of moments:
pressure : (75.6766483117 133.74306297 74.136257627)
viscous : (-0.0128092707646 9.11536978892 10.8799583077)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.0956427502562
Cd = 0.567786125923
Cl = -0.172335605448
Cl(f) = 0.0094749475324
Cl(r) = -0.18181055298
```

y-yosuke, 2016-07-20:
【メモ】OpenFOAM force
https://github.com/OpenFOAM/OpenFOAM-2.1.x/tree/master/src/postProcessing/functionObjects/forces

y-yosuke, 2016-07-20:
Cl(f) Cl(r) は Reference Length を使っているし車軸上の Clf, Clr とは違うもののような気がする．

他の計算が出来ているCFDの設定を見ても大体が計算しないような設定になっている．
わざわざそうしているように思える．

y-yosuke, 2016-07-20:
試行解析結果
Center of rotation ( 1.325, 0, -0.11 )
```
forces forces00 output:
sum of forces:
pressure : (196.29148201 -197.586939295 -65.3514275332)
viscous : (16.7948184932 -1.11385066373 0.675038074106)
porous : (0 0 0)
sum of moments:
pressure : (75.6766483117 133.74306297 74.136257627)
viscous : (-0.0128092707646 9.11536978892 10.8799583077)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.0956427502562
Cd = 0.567786125923
Cl = -0.172335605448
Cl(f) = 0.0094749475324
Cl(r) = -0.18181055298
```
Center of rotation ( 0, 0, -0.11 )
```
forces forces00 output:
sum of forces:
pressure : (194.408332311 -194.333708763 -63.7081602591)
viscous : (16.6478192914 -1.11914788935 0.644991542276)
porous : (0 0 0)
sum of moments:
pressure : (73.9624728462 215.377195539 -182.732826086)
viscous : (0.0157822643575 8.25131056396 9.30624304809)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.149717765667
Cd = 0.562376625753
Cl = -0.168037044943
Cl(f) = 0.065699243196
Cl(r) = -0.233736288139
```
Center of rotation ( 0, 0, 0 )
```
forces forces00 output:
sum of forces:
pressure : (185.099019095 -179.422279848 -48.0303545506)
viscous : (15.5493742288 -1.15507896112 0.675553213495)
porous : (0 0 0)
sum of moments:
pressure : (44.8134657614 154.725722842 -170.745071015)
viscous : (-0.121218749836 5.96482294569 8.51224508789)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.107581228791
Cd = 0.534644290362
Cl = -0.126180796849
Cl(f) = 0.0444908303664
Cl(r) = -0.170671627215
```

y-yosuke, 2016-07-20:
Result Control
- Center of rotation : ( 1.325, 0, -0.11 ) = ホイールベース中心
- Reference length value : 2.65 = ホイールベース

とすると force から手計算で大雑把に Fl, Fr を計算した結果と Cl(f), Cl(r) の結果が一致する．
バランスの値としては更に悪化するが．（下記囲み）

おそらくはこれが CFD で計算した中では正しい計算結果のように考える．
現状，風洞や実車との相関が取れないのでしょうがない．

その結果に沿うならば，フロントのダウンフォースが足りない．むしろリフトしている．
これはドラッグに比してダウンフォースが少ないためもあると思う．L/D 低過ぎ．
Cd はCd を落とす努力をしている一般乗用車に比べたら大きいが 0.56 ほどなのでフォーミュラだしそんなものかなと思う．

結果が実車と比べて真逆なら問題だが，程度の問題のうちに入るのであれば，
フロントのダウンフォースを稼ぐ手段は考えておいても損は無いように考える．
リアは増やす手段が比較的多く，リアウィングを上げるなどの明確な手段も残っているので．

とりあえずは，もう少しCFD解析結果の検証のためリアウィングが無いケースを計算させる予定．

```
forces forces00 output:
sum of forces:
pressure : (194.408332311 -194.333708763 -63.7081602591)
viscous : (16.6478192914 -1.11914788935 0.644991542276)
porous : (0 0 0)
sum of moments:
pressure : (73.9624728462 130.963883196 74.7593380255)
viscous : (0.0157822643575 9.10592435748 10.7891140015)
porous : (0 0 0)
forceCoeffs forceCoeffs01 output:
Cm = 0.14084061238
Cd = 0.562376625753
Cl = -0.168037044943
Cl(f) = 0.0568220899087
Cl(r) = -0.224859134851
```

y-yosuke, 2016-07-20:
風洞との相関をとるのは無理にしても他のCFDのソルバとの比較は出来ないものか？

y-yosuke, 2016-07-20:
【メモ】Clf 増大方法案

- [ ] Monocoque ドライバ足下の面を水平に
  - サスペンションジオメトリ，ペダル類の設計にもメリット有り
- [ ] Fr Wing - Fender の縦のつながりの部位外側にダクト
- [ ] Fr Wing Profile を厚くダウンフォースを発生しそうなものに
- [ ] Nose Cone S-Duct

y-yosuke, 2016-07-20:
その前にメッシュの粗さ加減問題もあるな．

[17 more replies]

closed 2017-06-29 by y-yosuke