y-yosuke/formula-ppoino #16: シミュレーションの内容を英語で説明

y-yosuke, 2016-09-09:
して下さい．とのことで英語編集用の Issue．ここで編集してHTMLでコピーかな．
英語でメールよりも長い（そんなに長くないけど）文章書くの久しぶり... orz

y-yosuke, 2016-09-09:
Editing now...

The Car

This car project is my formula style 2-seater car project published in GitHub.

https://github.com/y-yosuke/formula-ppoino
https://github.com/y-yosuke/ppoino-cars

"fp" are the initials of "formula-ppoino". "-ppoino" is a Japanese expression that means "a thing(s) looks/sounds like -", so "formula-ppoino" means "a car looks like formula".

I designed the car in the style of the early 2000s formula cars.
The reason of "2-seater" is to feel the joy riding in the formula car with friends on the road, it is not only for the competitions in racing circuits.

The model

I designed the car using Rhinoceros.

For importing multiple faces body, front wheels and rear wheels into one geometry, I export ASCII STL files of the each faces from Rhinoceros and edit the names of the solids in these ASCII STL files manually. Then I put these STL files together into the one STL file using "cat" command and upload it.

The size of the mesh is required 9M nodes at least in this case. It was derived from the result of the simulations in "fp-023b Test-C".
https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/

The aim of the simulations

front and rear downforce balance

y-yosuke, 2016-09-09:
Editing now...

## The aim of the simulations

Generally in the aerodynamic development of the racing cars, you mainly focus on the characteristics below.

- Downforce, Drag, L/D
- Balance of the front and rear downforces
- Sensitivity of the Downforce and Balance to the vehicle ride heights

Through the simulations of ""fp-023b Test-C" https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/", I found the front downforce was not enough. It depends on the car weight balance and how the vehicle ride heights change, but I provisionaly set the goal that the center of the front and rear downforces will be about 60-65%. ( Clf : Clr = 40-35 : 60-65 )

emijah, 2016-09-09:
Edited

## The aim of the simulations

Generally, during aerodynamic development of racing cars, you mainly focus on the characteristics below.

- Downforce, Drag, L/D
- Balance of the front and rear downforces
- Sensitivity of the Downforce and Balance to the vehicle ride heights

Through simulation, as in ""fp-023b Test-C" https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/", I found there was insufficient front downforce. It depends on the car weight balance and how the vehicle ride heights change, but I provisionaly set the goal to distribute the front and rear downforces to be about 60-65%. ( Clf : Clr = 40-35 : 60-65 )

y-yosuke, 2016-09-09:
@emijah ありがとうございます！

y-yosuke, 2016-09-09:
Editing...

Through simulations, as in "fp-023b - Front Aero Devices https://www.simscale.com/projects/yosukegb4/fp-023b_control/ ",

emijah, 2016-09-09:
@y-yosuke これよりも前の部分は直接投稿した文書を編集してしまったかもしれません。確認しますね。

emijah, 2016-09-09:
Edited

## The car

This car project is my formula style 2-seater car project published in GitHub.

https://github.com/y-yosuke/formula-ppoino

I designed the car in the style of the early 2000s formula cars. The reason it's a "2-seater" is to allow you to share the joy of riding in a formula car with your friends on the road, it is not only for competition on racing circuits.

The prefix "fp" are the initials of "formula-ppoino". "-ppoino" is a Japanese expression that means "thing(s) looks/sounds/drives like -", so "formula-ppoino" means it "looks like a formula car".

## The model

I designed the car using the modelling software Rhinoceros.

To import multiple faces, the body, front wheels and rear wheels into one geometry, I exported ASCII STL files of each faces from Rhinoceros and edited the names of the solids in these ASCII STL files manually. Then I put these STL files together into the one STL file using "cat" command and uploaded it.

The size of the mesh required is at least 9M nodes in this case. It was derived from the result of the simulations in "fp-023b Test-C".
https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/

y-yosuke, 2016-09-09:
ちなみに SimScale の中の人から説明を書き加えて下さいというお願いで，
下記フォーラムのページを参考にここまでは全然詳しくなくて大丈夫とのこと．
https://www.simscale.com/forum/t/wind-turbine-blade-fluid-and-structural-optimisation-using-cae/37492

主に私が SimScale に貼った ParaView で Post-process した画像の解説が欲しいそうです．
（それらの説明がそれなりであれば）Project Spotlight として来週あたり紹介するかもとのこと．

emijah, 2016-09-09:
> optimisation-using-cae/37492

よかったねぇ！
実際につくラメのスポンサーも募集中と書いてみては？

y-yosuke, 2016-09-09:
GitHub のコメントって編集履歴は残らないんでしたっけ？

y-yosuke, 2016-09-09:
## The result of "fp-023b - Front Aero Devices" (Summary)

- S-Duct
- Front Wing t40/t50 Comparison
- Front Fender Narrow Pillar
- Front Inner Fender Modification
- Front Flap
  - Position X+20mm
  - Angle +7.5deg (mechanical MAX +15deg)

## The result of "fp-023c - Front Aero Devices"

y-yosuke, 2016-09-10:
Editing...

## fp-023c - Front Aero Devices

y-yosuke, 2016-09-10:
【 fp-023c - Summary of the Results (Part 2) 】Front Fender Stay (Part 1)

Edited in https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246096612

y-yosuke, 2016-09-10:
【 fp-023c - Summary of the Results (Part 3) 】Front Fender Stay (Part 2)

Edited in https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246107120

y-yosuke, 2016-09-10:
後は...

- [ ] S-Duct B の比較
- [ ] Ride height の比較

かな．
渦度に関しては私の工学的な理解が浅いので説明はしないでおこうかと思っている．

y-yosuke, 2016-09-11:
【 fp-023c - Summary of the Results (Part 4) 】S-Duct B

Edited in https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246163741

y-yosuke, 2016-09-11:
Front / Rear のダウンフォースの割合は風洞実験の経験から設定した暫定値なので
本シミュレーション上ではそれは Body Only の Coefficients に相当するから "of the body" を入れた方が良い気がする．

> I provisionaly set the goal to distribute the front and rear downforces to be about 60-65%.

I provisionaly set the goal to distribute the front and rear downforces of the body to be about 60-65%.

y-yosuke, 2016-09-11:
"
Ride Heights Comparison

Ideally, various vehicle ride heights are tested for each aero-devices to know the sensitivity of the downforce and balance to the ride heights. But it needs a lot of 'core hours' to solve.
"

y-yosuke, 2016-09-11:
【 fp-023c - Summary of the Results (Part 5) 】Front Fender Louver C + Louver Stay B

Edited in https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246167618

y-yosuke, 2016-09-11:
【 fp-023c - Summary of the Results (Part 6) 】Ride Heights Comparison

Edited in https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246168325

y-yosuke, 2016-09-11:
私の知識で説明出来るのは大体これら ↑ ぐらいかな．

やはり乱流エネルギー k の図は説明に十分な知識を持ち合わせていないので避けておこう．

y-yosuke, 2016-09-12:
修正していただいたものをまとめ，あんど，投稿．

## The car

This car project is my formula style 2-seater car project published in GitHub.

https://github.com/y-yosuke/formula-ppoino

I designed the car in the style of the early 2000s formula cars. The reason it's a "2-seater" is to allow you to share the joy of riding in a formula car with your friends on the road, it is not only for competition on racing circuits.

The prefix "fp" are the initials of "formula-ppoino". "-ppoino" is a Japanese expression that means "thing(s) looks/sounds/drives like -", so "formula-ppoino" means it "looks like a formula car".

## The model

I designed the car using the modelling software Rhinoceros.

To import multiple faces, the body, front wheels and rear wheels into one geometry, I exported ASCII STL files of each faces from Rhinoceros and edited the names of the solids in these ASCII STL files manually. Then I put these STL files together into the one STL file using "cat" command and uploaded it.

The size of the mesh required is at least 9M nodes in this case. It was derived from the result of the simulations in "fp-023b Test-C".
https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/

## The aim of the simulations

Generally, during aerodynamic development of racing cars, you mainly focus on the characteristics below.

- Downforce, Drag, L/D
- Balance of the front and rear downforces
- Sensitivity of the Downforce and Balance to the vehicle ride heights

Through simulation, as in ""fp-023b Test-C" https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/", I found there was insufficient front downforce. It depends on the car weight balance and how the vehicle ride heights change, but I provisionaly set the goal to distribute the front and rear downforces of the body to be about 60-65%. ( Clf : Clr = 40-35 : 60-65 )

## The conditions of the simulations
The conditions are in accordance with the state of the car running on public roads.

- Inlet velocity, Moving floor speed : 30 m/s (= 108 km/h)
- Ride height : Front: 110mm / Rear: 110mm (basically)

## The result of "fp-023b - Front Aero Devices" (Summary)

I tested some front aero-devices in the simulations "fp-023b Front Aero Devices https://www.simscale.com/projects/yosukegb4/fp-023b_control/ ". The following front aero-devices increased the front downforce effectively. These decreased the aerodynamical front lift to almost zero, but the ratio of front-rear downforce was still 96% on the rear axle.

- S-Duct at Nose Cone
- Front Wing t50 (changed from t40)
- Modified Front Inner Fender
- Front Flap
  - Position X+20mm
    - Angle +7.5deg (mechanical MAX +15deg)

Coefficients of the Body and Wheels							

Coef.	|	fp-023b Control	|	fp-023c Control	|	Difference	|	Difference %	|	Remarks
---	|	---	|	---	|	---	|	---	|	---
Cm	|	0.1330	|	0.0781	|	-0.0549	|	-41.3	|	
Cd	|	0.5510	|	0.5590	|	+0.0080	|	+1.4	|	
Cl	|	-0.1550	|	-0.1693	|	-0.0142	|	-9.2	|	
Clf	|	0.0555	|	-0.0066	|	-0.0620	|	-111.8	|	
Clr	|	-0.2105	|	-0.1627	|	+0.0478	|	+22.7	|	
CoP	|	1.3579	|	0.9613	|	-0.3966	|	-39.7	|	
L/D	|	-0.2813	|	-0.3028	|	-0.0215	|	-7.6	|	

- fp-023b Control : Base of "fp-023b - Front Aero Devices"
- fp-023c Control : Best result of "fp-023b - Front Aero Devices"

y-yosuke, 2016-09-12:
各項目題の "The" は外すかな...

emijah, 2016-09-12:
今日の昼休みにまた見ますね

y-yosuke, 2016-09-12:
ありがとうございます．正午までに整えてまとめておきます．

y-yosuke, 2016-09-12:
## Front Fender Louvers

## Front Fender Stay (Part 1)

## Front Fender Stay (Part 2)

## S-Duct B

## Front Fender Louver C + Louver Stay B

## Ride Heights Comparison

y-yosuke, 2016-09-12:
STL モデルのエクスポート・インポートの話は別に重要じゃないかな... とも思うけど，
私の方からは投稿はしておいて，先方で不必要だと思えば転載のときに削ってもらえるでしょう．

y-yosuke, 2016-09-12:
Conclusion みたいなのもあった方が良いのかな．

emijah, 2016-09-12:
ルーバーってlouvreって綴りじゃなかったでしたっけ？

y-yosuke, 2016-09-12:
https://en.wikipedia.org/wiki/Louver
Louver : 米国語
Louvre : 英国語
のようです．レーシングカーでは「英国語」が多い気がしますが，
会社員当時は適当に Louver を使っていたような気がします．

y-yosuke, 2016-09-12:
"Musée du Louvre" も Louvre か．勉強になるー

emijah, 2016-09-12:
意図的にLouvreを広めていたつもりだったのですが、違ったか。
Conclusionsを現在進行形にするかどうかですが、今やっている内容でとどめるならば、それでいいんじゃないですかね？

一度まとまったやつをEdited

## Contents

- Car
- Model
- Conditions
- Aim of the simulation
- Summary of simulation "fp-023b - Front Aero Devices"
                       ^^^
- Front Fender Louvers
- Front Fender Stay (Part 1)
- Front Fender Stay (Part 2)
- S-Duct B
- Front Fender Louver C + Louver Stay B
- Ride Heights Comparison
- Conclusions

## Car

This car project is my formula style 2-seater car project published in GitHub.

https://github.com/y-yosuke/formula-ppoino

I designed the car in the style of the early 2000s formula cars. The reason it's a "2-seater" is to allow you to share the joy of riding in a formula car with your friends on the road, it is not only for competition on racing circuits.

The prefix "fp" are the initials of "formula-ppoino". "-ppoino" is a Japanese expression that means "thing(s) looks/sounds/drives like -", so "formula-ppoino" means it "looks like a formula car".

## Model

I designed the car using the modelling software Rhinoceros.

To import multiple faces, the body, front wheels and rear wheels into one geometry, I exported ASCII STL files of each faces from Rhinoceros and edited the names of the solids in these ASCII STL files manually. Then I put these STL files together into the one STL file using "cat" command and uploaded it.

The size of the mesh required is at least 9M nodes in this case. It was derived from the result of the simulations in "fp-023b Test-C".
https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/

## Conditions

The conditions of the simulations are in accordance with the state of the car running on public roads.

- Inlet velocity, Moving floor speed : 30 m/s (= 108 km/h)
- Ride height : Front: 110mm / Rear: 110mm (basically)

## Aim of the simulations

Generally, during aerodynamic development of racing cars, you mainly focus on the characteristics below.

- Downforce, Drag, L/D
- Balance of the front and rear downforces
- Sensitivity of the Downforce and Balance to the vehicle ride heights

Through simulation, as in "fp-023b Test-C https://www.simscale.com/projects/yosukegb4/fp-023b_test-c/ ", 
                                  ^
I found there was insufficient front downforce. It depends on the car weight balance and how the vehicle ride heights change, but I provisionaly set a goal to distribute the front and rear downforces of the body to 
                                                                            ^変わるから
be about 60-65%. ( Clf : Clr = 40-35 : 60-65 )

## Summary of of the simulation "fp-023b - Front Aero Devices"

I tested some front aero-devices in the simulation "fp-023b - Front Aero Devices https://www.simscale.com/projects/yosukegb4/fp-023b_control/ ". The following front aero-devices increased the front downforce effectively. These decreased the aerodynamical front lift to almost zero, but the ratio of front-rear downforce was still 96% on the rear axle.

- S-Duct at Nose Cone
- Front Wing t50 (changed from t40)
- Modified Front Inner Fender
- Front Flap
  - Position X+20mm
    - Angle +7.5deg (mechanical MAX +15deg)

Coefficients of the Body and Wheels							

Coef.	|	fp-023b Control	|	fp-023c Control	|	Difference	|	Difference %	|	Remarks
---	|	---	|	---	|	---	|	---	|	---
Cm	|	0.1330	|	0.0781	|	-0.0549	|	-41.3	|	
Cd	|	0.5510	|	0.5590	|	+0.0080	|	+1.4	|	
Cl	|	-0.1550	|	-0.1693	|	-0.0142	|	-9.2	|	
Clf	|	0.0555	|	-0.0066	|	-0.0620	|	-111.8	|	
Clr	|	-0.2105	|	-0.1627	|	+0.0478	|	+22.7	|	
CoP	|	1.3579	|	0.9613	|	-0.3966	|	-39.7	|	
L/D	|	-0.2813	|	-0.3028	|	-0.0215	|	-7.6	|	

- fp-023b Control : Base of "fp-023b - Front Aero Devices"
- fp-023c Control : Best result of "fp-023b - Front Aero Devices"

emijah, 2016-09-12:
そうそう。キーワードをlouvre aerodynamicにしないと永久にリファレンスは出てきません

y-yosuke, 2016-09-12:
シミュレーションの結果のまとめ各項目もこっちにコピーしておいた方が良かったですか？

Front Fender Louvers
Content here : #12 (comment)

y-yosuke, 2016-09-12:
前半の修正点 ^ 理解・了解しました．

emijah, 2016-09-12:
Edited

## Front Fender Louvres

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246084217

## Front Fender Stay (Part 1)

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246096612

## Front Fender Stay (Part 2)

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246107120

## S-Duct B

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246163741

## Front Fender Louvre C + Louvre Stay B

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246167618

## Ride Heights Comparison

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246168325

## Conclusions

Conclusions or Conclusion ???

Content here : https://github.com/y-yosuke/formula-ppoino/issues/12#issuecomment-246227805

emijah, 2016-09-12:
動画とか見入っちゃいますね

y-yosuke, 2016-09-12:
これはリンク先の英語の内容も大丈夫だったということでしょうか？

y-yosuke, 2016-09-12:
SimScale の forum は基本的には Markdown に対応しているものの Markdown の表には対応していない HTML でないとだめのようだ．

emijah, 2016-09-12:
リンク先は後で確認します

y-yosuke, 2016-09-12:
すみません．よろしくお願い致します．

y-yosuke, 2016-09-12:
HTML でも’表’が表示されないなぁ...

y-yosuke, 2016-09-12:
表は TeX で書きなはれ，ということなのかな？
https://www.simscale.com/forum/t/mathjax-arrays/444

y-yosuke, 2016-09-12:
TeX でも表示されない．キャプチャして画像貼りつけかな．

y-yosuke, 2016-09-12:
Louver / Louvre 問題は既に ParaView キャプチャした画像に Louver で書いてあるので Louver で統一します．

y-yosuke, 2016-09-12:
もしかしてこれらのドキュメント全体の標題も必要な気がしてきた...
Result of "fp-023c Front Aero Devices" ? Results ???

emijah, 2016-09-12:
@y-yosuke なにー！(Louvre)

fp-023c - Front Aero Device Study

はどうですか？

y-yosuke, 2016-09-12:
@emijah すみません (Louvre)

標題はそれにしましょう．

y-yosuke, 2016-09-12:
アップロード始めました．"Sorry. Now Editing..." を冒頭に書きながら．
追加修正点は私が後から手動修正します．
https://www.simscale.com/forum/t/fp-023c-control-simulation-project-by-yosukegb4/36689/17

y-yosuke, 2016-09-13:
説明用にアップロードした画像をこちらにも保管しておく．
[image]

y-yosuke, 2016-09-13:
（図を多く貼るとボリュームが出て沢山書いたような気分になる）

y-yosuke, 2016-09-13:
全体の項目の階層を次のようにして投稿済みとしました．
@emijah さん，協力いただき大変ありがとうございました．

### Contents

- Introductions
  - Car
  - Model
  - Conditions
  - Aim of the simulation
  - Summary of simulation "fp-023b - Front Aero Devices"
- Simulations
  - Front Fender Louvers
  - Front Fender Stay (Part 1)
  - Front Fender Stay (Part 2)
  - S-Duct B
  - Front Fender Louver C + Louver Stay B
  - Ride Height Comparison
  - Conclusion

y-yosuke, 2016-09-13:
（あれもこれも説明しなきゃと思って of とか the とかいろいろと増殖して格好悪い表現になっていく私の英語力... orz）

y-yosuke, 2016-09-14:
無事に紹介されました．URLもそれらしいものに変えて頂いたようです．
https://www.simscale.com/forum/t/fp-023c-front-aero-device-study/36689

あらためて @emijah さん協力ありがとうございました．

ということで，この Issue は Close します．

closed 2016-09-14 by y-yosuke

y-yosuke, 2016-09-14:
（そういえば SimScale は OnShape や Autodesk と連携しているから Rhinoceros に関係した記述も編集されずにそのままだったな... ）

emijah, 2016-09-14:
連携ができているということはツールとしては大きなことですからね。中途半端な支援しかできずもうしわけないです。

y-yosuke, 2016-09-14:
いえいえ，修正して頂いた点から英語の感覚が少し分って良かったです．

OnShape / Autodesk はテクニカルパートナーなのかな？と勝手に思っていましたが
いらぬ心配だったようです．