y-yosuke/formula-ppoino #20: Small Radio Control Car

y-yosuke, 2017-03-08:
- Bluetooth Controller - Vagabond Works : bCore Series
  - http://vagabondworks.jp/blog-category-25.html
  - RCミニ四駆プロジェクト
    - https://kibidango.com/425
- Linear Servo
  - HISKY HFP100 WLTOYS V944 RC Helicopter Parts LS91 Linear Servo
    - http://amzn.asia/3VaiuBY
  - HISKY HFP100 WLTOYS V944 RC Helicopter Parts LS90 Linear Servo
    - http://amzn.asia/3gVufaX
- Micro RC Truck (Video)
  - https://www.shapeways.com/forum/t/micro-rc-truck-video.21355/

y-yosuke, 2017-03-08:
Indoor Airplane World
http://indoor-airplane-world.com/

y-yosuke, 2017-03-08:
Hirosugi - Shaft
http://hirosugi.co.jp/products/U/SRU-0000.html#bx1

y-yosuke, 2017-03-08:
N gauge (N scale) warm gear
http://www5a.biglobe.ne.jp/~toyoyasu/m120115.htm

emijah, 2017-03-08:
いろいろとありますね！
なにか動く形のものになると楽しそう．
もう少し加勢できるいいのですが．

y-yosuke, 2017-03-08:
サーボ
https://hobbyking.com/jp_jp/hobbyking-ultra-micro-servo-1-7g-for-3d-flight-right.html
https://hobbyking.com/jp_jp/hobbyking-ultra-micro-servo-1-7g-for-3d-flight-left.html

y-yosuke, 2017-03-11:
ふむ．京商 スライドサーボ

[image]

[image]