y-yosuke/formula-ppoino #22: Round Shape Display on Steering Wheel

y-yosuke, 2017-03-26:
ステアリングホイール設置のディスプレイを
円形のディスプレイにしてステアリングホイールを回しても表示をその分逆回しにして
見かけ上回らないようにするものがないか？ できないか？

y-yosuke, 2017-03-26:
シャープ、円形液晶も実現可能な「フリーフォームディスプレイ」開発。IGZO応用。ウェアラブルにも
http://av.watch.impress.co.jp/docs/news/653959.html

シャープは今どうなっているのだろう？

y-yosuke, 2017-03-26:
出来るだろうけど，簡単に調達可能か？というところか．直径 120mm ぐらい．

y-yosuke, 2017-03-26:
四角のディスプレイでも丸枠をつけて，はみ出した部分は別窓ディスプレイとして利用すれば良いか．

y-yosuke, 2017-03-26:
下のスクショで内側の丸が直径 116mm なので使えそうな大きさ．
[image]

y-yosuke, 2017-03-27:
円と四角
[image]
[image]

emijah, 2017-03-28:
丸い方が形がいいですね。

y-yosuke, 2017-03-28:
四角はF1のようなスパルタンな感じがしますね．
丸はカジュアルな感じで公道を楽しむクルマのコンセプトには合っているようにも思います．

丸の場合は表示はクラシカルなアナログメータ表示モードがあると良いかも．

emijah, 2017-03-29:
ってことは、用途により使い分けを促せるんですね。
四角いディスプレイの場合はレースモードになって、パワー重視セッティング。丸いディスプレイは峠で楽しむレベルの距離重視セッテイングになるというのもいいですね。

y-yosuke, 2017-03-30:
円形はそれに見合ったディスプレイがリーズナブルに調達可能な場合に採用というところでしょうか．

CANの仕様等私は分っていないのですが，CAN通信をする市販のステアリングホイールがあれば
ソフトウェアの調整等は必要としても，それと互換可能にすることはできるのかな？

y-yosuke, 2017-04-07:
速度とか加速度などはパッセンジャーと共有した方が楽しそうだし，
センターコンソールのディスプレイに表示した方が良いように思う．

y-yosuke, 2017-04-20:
https://github.com/y-yosuke/ppoino-cars/issues/8#issuecomment-295570767

emijah, 2017-04-20:
OBD通信が読めるのかどうかというところでしょうかね？

y-yosuke, 2021-03-22:
ホンダのコンセプトカー？ でステアリング中心に丸形のディスプレイを設けて，ステアリングホイールを回しても表示が回らないようにしている動画がありました．
https://youtu.be/DT47g-OvCGg?t=92

y-yosuke, 2021-04-14:
Koenigsegg Gemera はステアリングホイール軸上ではないけど液晶ディスプレイ表示を回転させて水平にしている．

The $2m Koenigsegg Gemera That I Chased AROUND THE WORLD!
https://youtu.be/Y3qgpbuw9f0?t=621

emijah, 2021-04-15:
> ステアリングホイール設置のディスプレイを
> 円形のディスプレイにしてステアリングホイールを回しても表示をその分逆回しにして
> 見かけ上回らないようにするものがないか？ できないか？

解像度にかなり余裕が必要そう。でも描いて確認することはできますね

y-yosuke, 2021-06-22:
ベントレー、ゲームと共用できるステアリングホイール開発…パイクスピーク2021参戦車に装着https://response.jp/article/img/2021/06/22/346975/1645636.html

y-yosuke, 2022-03-27:
Koenigsegg のステアリングメータパネルも水平保持式

- Ultimate Koenigsegg Factory Tour With Christian Von Koenigsegg
  - https://youtu.be/30TFrquvs9g?t=1125

y-yosuke, 2022-04-01:
- 1080P Round Shape LCD Video Display 5” (130mm) Diameter
  - https://www.youtube.com/watch?v=M7b6qAsA16k

emijah, 2022-04-02:
うーーーーん、いつまで作っているか、ですね。一般的に使われる用途があればいいのですが。
掛け時計か置き時計としてsmart clockが普及すると、心配せずに使えそうですね

y-yosuke, 2022-04-02:
四角いディスプレイでも丸くフレームを作ってあとはステアリングホイール構造物のようなフリして隠してしまえばできそうですけどね．

y-yosuke, 2022-04-02:
こういう ↓ 需要が続けば継続的に丸形も生産されるかな．

- Cammus StreetSport 60mm i-Round LCD Gauge Multifunction
  - https://shop.windbooster.com/product/60mm-round-lcd-gauges/
  - https://cammus.com/streetsport-lcd-meter/

emijah, 2022-04-03:
Ikea, ニトリの時計が無敵だと思います。
一般車車の計器でも良いですね

y-yosuke, 2022-05-27:
- Bentley GT3 Fanatec Steering Wheel
  - https://bentley.fanatec.com/