y-yosuke/formula-ppoino #24: Fr Suspension

y-yosuke, 2017-05-07:
#3

y-yosuke, 2017-05-07:
Fr camber angle about 10deg

- Bump Steer
  - Stroke Z+50mm : 0.0493 Toe Out
  - Stroke Z-50mm : 0.0202 Toe Out

y-yosuke, 2017-05-16:
第9回 chapter3 サスペンション・デザイン
その5:ステアリング・ジオメトリーとその検討
https://www.zf.com/japan/ja_jp/corporate/motorsports_1/webseminar/9chapter3/webseminarvol9chapter3.html
"少なくとも走行時に起点となるサスペンション・ストロークから縮み・伸びのストロークが始まる範囲では、トー変化が出ない設計としたい。さらにストロークが深くなったところでトー変化が生じざるをえない場合は、バンプ→トーアウト(タイロッド前方配置ならナックルアーム端が外側へ、後方配置なら逆に内側へ)にしたほうがドライビングの対応は多少楽になる。"

y-yosuke, 2017-05-16:
ステアリングラックをもう少し高い位置にしたら良いのかな？

emijah, 2017-05-16:
う、フロントサスのレイアウトが頭に入っていない..

y-yosuke, 2017-05-16:
@emijah あ，「？」を気にしないで下さい．自分への問いが多いので．

y-yosuke, 2017-05-16:
ステアリングラックを 10mm 高くした場合

- Stroke Z+50mm : 0.46 deg Toe Out
- Stroke Z-50mm : 0.71 deg Toe In

桁違いに角度が大きくなる．
Toe In/Out 逆だけど元の位置の方が絶対的に角度が小さいし良いような気がする．

y-yosuke, 2017-05-16:
今のサスペンションレイアウト（線図）
[image]

y-yosuke, 2017-05-16:
分りにくいので四面図
[image]

y-yosuke, 2017-05-16:
部品を表示したキャプチャ
[image]

y-yosuke, 2017-05-17:
Parts Assembly
[image]

y-yosuke, 2017-05-17:
（機能的にという意味ではなくて値段的に高級サスペンションになってしまいそうな...）

y-yosuke, 2017-06-15:
Spherical / Rodend Angle

- Upr Wishbone Inner : 
- Lwr Wishbone Inner : 9.1389 deg

y-yosuke, 2017-06-15:
Spherical MBW-15VCR for Front Lower Wishbone Inner Points

y-yosuke, 2017-06-15:
Rodend HR15E = angle 11 deg
+ Dia.18 mm Collars (or Dia. 19mm with Chamfer 0.5mm) -> angle 12.5 deg

y-yosuke, 2026-09-14:
キャスター角の見直しのついでにその他の要素を検討していてストロークに伴うロールセンター変化が大きかったので Lower Wishbone のセンター締結をやめてモノコックの両端に近いポイントにしようと思う．

[image]
[image]

y-yosuke, 2026-09-14:
ちなみに再設計前のロールセンターの移動範囲はこんな感じ．

[image]

y-yosuke, 2026-09-16:
ミネベアミツミのロッドエンド RBT シリーズが廃番？になったようなので代替品を探す．

- 株式会社エムティアール - 標準品
  - https://mtr-miyamoto.jp/normal

y-yosuke, 2026-09-17:
- BlueFast® metric hexagonal K-nuts
  - https://bluefast.com/metric-hexagonal-k-nuts-datasheet/
  - M3 〜 M18
    - BFK HEXA M16150* M16 x 1,5 = * special sizes on request
- Demon Tweeks Metric Aerospace Locknut (K-Nut)
  - https://www.demon-tweeks.com/uk/demon-tweeks-metric-aerospace-locknut-k-nut-243560/
  - M3 〜 M14
    - M16 は無かった...