y-yosuke/formula-ppoino #26: Cockpit Console

y-yosuke, 2017-05-16:
Rear View Monitor Layout

8inch ResizingLCD 300nits LED BackLight / 1024×324　215×79.2×7.5
http://www.skr-tech.co.jp/okugailcdkanban.htm

Driver's perspective
[image]

y-yosuke, 2017-05-16:
RH/LH コンパチブルにするためにセンターに置きたいところだけど...
無理せずに助手席側にずらすか...

y-yosuke, 2017-05-17:
いや，バックビューモニターがそもそも不要か．

emijah, 2017-05-17:
不要に一票

y-yosuke, 2017-05-17:
@emijah ですよね．

カーナビなどと同じで欲しければ勝手に付けて下さいで済みますね．
左ハンドル用のステアリングシャフトの締結点もモノコック側に余るだろうし Aux Power ぐらい出すし．

y-yosuke, 2017-05-17:
Ariel Atom Interior
https://www.autocar.co.uk/car-review/ariel/atom/interior

y-yosuke, 2017-10-11:
Cockpit Panel and Bulkhead
[image]

y-yosuke, 2017-10-12:
肩部の幅が 510mm ぐらいなので
バケットシートに座った状態の肩幅がそれを超える場合は痩せてください．

y-yosuke, 2017-10-13:
- Cockpit Interfeces
  - Cut Off
  - Fire
  - Hazard Lamp
  - Shift Switch
    - Drive
      - Insane
      - Normal
      - Eco / Long Drive
    - Neutral
    - Reverse
    - Park
  - Parking Brake (Electrical Switch?)
  - Turn
    - Left
    - Right
  - Head Light
    - High
    - Low
  - Horn
  - Braking Ratio
  - EV Brake Mode Selector

and anything else?

emijah, 2017-10-15:
InsaneはTeslaの匂いがちょっとしますね。もう少し違った表現がないですかね？

y-yosuke, 2017-10-15:
Ridiculous Mode, Ludicrous Mode もテスラが使ってるっぽいですね．
日本語だったら「加速番長」

y-yosuke, 2017-10-15:
Dual Motor で Software Differencial だったらその調整インタフェースも欲しいか．
可能な限りインタフェースはメカニカルにしたい．

emijah, 2017-10-18:
「加速番長」いいですね。
「交差点GP」ボタンにするといろいろと問題が起きそうですね。

y-yosuke, 2017-10-22:
絶対NG「煽りモード」

emijah, 2017-11-17:
人名が出てこなければ一般名詞だから...

y-yosuke, 2017-12-11:
Cockpit Panels
[image]

y-yosuke, 2018-01-29:
サイドミラーの大きさは
自動車用もバイク用もいわゆる「新保安基準」適合であれば同じで良いと読めるけどあってるかな？
http://www.mlit.go.jp/common/000187515.pdf

emijah, 2018-01-29:
実際に相談するしかなさそうですね。でもミニみたいにデザインを優先して見にくいミラーもあるので、意外と自由なのかもしれないですね。

y-yosuke, 2018-01-29:
BMWのミニは画像を検索している限りでは保安基準に適合してそうに見えます．

y-yosuke, 2018-01-29:
むしろ適合するギリギリのサイズに合わせると 直径78mm の円を含む 69cm^2 以上の面積から
左右を詰めたような変な大きさになります．

emijah, 2018-01-29:
意外と小さいですね。

y-yosuke, 2018-01-29:
そうなんですよね．一般の乗用車としては小さいのでそれで良いのかな？と引っかかりを覚えました．

家の車はむしろ最大の大きさ 200mm x 120mm にギリギリ収まるような大きさっぽいので
運転上良好な視界を確保するにはそっちに合わせるのかもしれません．

formula-ppoino では空力的にサイドミラーは悪さをしているので
なるべく小さく，なるべく悪影響の少ない形状にしたいなと思案しているところです．

ミラーレス・カメラ使用が理想的なのかもしれないのですが
コクピットが小さくてドライバサイドのモニターを設置しても
ステアリングホイールや腕に隠れて見えないので難しい感じです．

emijah, 2018-01-30:
ステアリングの斜め上辺りも無理そうですかね？

y-yosuke, 2018-01-30:
見える部分はあるにはあると思いますが，ステアリングを切ったときに腕で見えなくなりそうです．

y-yosuke, 2021-04-02:
自動車の後方確認カメラ、新車の装備義務化　来年５月以降にも - SankeiBiz
http://www.sankeibiz.jp/business/news/210401/bsc2104012245015-n1.htm

y-yosuke, 2022-03-18:
Morgan Super 3 のインテリア

- Shmee150 - I'm BUYING a MORGAN SUPER 3! New Addition to the Shmeemobiles
  - https://www.youtube.com/watch?v=sYndVOxAfKY

emijah, 2022-03-20:
確かにいい！真ん中だから右ハンドルも左ハンドルも同じにできそう。
横にある荷物を入れるバッグのバイクっぽい手軽さも好きです

y-yosuke, 2022-03-20:
可能な限り左右ハンドルは共通設計にしたいですね．

移動用の荷物はハード目のルーフトップキャリアにしてマクロスのストライクバルキリーみたいなマッチョデザインも良いかと思っていましたが，ソフト目のバッグを引っ掛けるのも良いですね．

＜参考＞
- VF-1S ストライクバルキリー ファイター 一条輝 機
  - https://www.hobby-wave.com/products/mc064/

y-yosuke, 2022-03-27:
Aston Martin Valkyrie は 2 dgrees シートが内向きになっていると聞こえるが，英語リスニング怪しい私．
ちなみに formula-ppoino は 3 degrees ．

- WORLD’S FASTEST Convertible! New Aston Martin Valkyrie Spider
  - https://youtu.be/52U1gV_7n6U?t=474

y-yosuke, 2022-03-27:
字幕つけたら良いだけだった． https://youtu.be/52U1gV_7n6U?t=480

y-yosuke, 2022-06-01:
Pedal position adjuster

- FIRST LOOK: Mercedes-AMG One - 1063hp Hypercar With F1 Engine Is FINALLY Finished! | Top Gear
  - https://youtu.be/Lyl-Yt7PBPo?t=846

y-yosuke, 2022-08-18:
- 自動車のタッチパネル操作は物理ボタンの4倍時間がかかる
  - https://gigazine.net/news/20220818-car-touch-screen-physical-button/

y-yosuke, 2022-11-17:
Mclaren Elva はヘルメット収納がついている．（スクーターみたい）

- World First: McLaren Elva - Detailing & Protection!
  - https://youtu.be/vj9rTc5tBEk?t=1262

y-yosuke, 2023-06-15:
- 3.4インチ円形タッチスクリーンディスプレイ（800 x 800, MIPI DSI接続) - SwitchScience
  - https://www.switch-science.com/products/9028
  - "有効表示サイズ：直径3.4インチ（87.6 mm）"

y-yosuke, 2023-08-14:
- 2.1-Inch Round Display with Magnetic Encoder Knob
  - https://www.youtube.com/watch?v=5jDzeOtMlGc

y-yosuke, 2026-09-25:
Neutral body posture に着座姿勢を近づけるか．

- Neutral body posture
  - https://en.wikipedia.org/wiki/Neutral_body_posture