y-yosuke/formula-ppoino #27: Rr Suspension

y-yosuke, 2017-05-18:
- Bump Steer Angle
  - Stroke Z+50mm : 0.538 deg Toe In
  - Stroke Z-50mm : 0.6817 deg Toe Out

大きすぎるな．

y-yosuke, 2017-05-18:
Track Rod Outer Joint on Upright +40mm ( Horizontal Track Rod )

- Bump Steer Angle
  - Stroke Z+50mm : 
  - Stroke Z-50mm :

y-yosuke, 2017-05-18:
Track Rod Outer Joint on Upright Origin +20mm

- Bump Steer Angle
  - Stroke Z+50mm :

y-yosuke, 2017-05-18:
Track Rod Outer Joint on Upright Z+25mm from Origin

- Bump Steer Angle
  - Stroke Z+50mm : 0.0936 deg Toe Out
  - Stroke Z-50mm : 0.0782 deg Toe In

y-yosuke, 2017-05-18:
Track Rod Outer Joint on Upright Z+22mm from Origin

- Bump Steer Angle
  - Stroke Z+50mm : 0.0160 deg Toe Out
  - Stroke Z-50mm : 0.0153 deg Toe Out

Best?

emijah, 2017-05-18:
おおおお

y-yosuke, 2017-06-13:
Geometry Plan New-B : Upper Wishbone / Upright Point =  20mm Inward

- Bump Steer Angle
  - Stroke Z+50mm : 0.0096 deg Toe In
  - Stroke Z-50mm : 0.0095 deg Toe Out

Ice Color : Recent Origin
Black Color : Plan New-B
[image]

Recent Origin seems better in the roll center ( Hign and Narrow ).

y-yosuke, 2017-06-13:
Geometry Plan New-C : Upper Wishbone / Upright Point = 20mm outer from Recent Origin

- Bump Steer
  - Stroke Z+50mm : 0.0396 deg Toe Out
  -  Stroke Z-50mm : 0.0202 deg Toe Out

Ice Color : Recent Origin
Lavender Color : Plan New-C
[image]

Recent Origin is the best in the roll center (Narrow Movement).

y-yosuke, 2017-06-13:
（なんか「たまたまそこがツボ」的に Recent Origin Geometry がベスト...）

y-yosuke, 2017-06-13:
or... is it just a local optimum?

y-yosuke, 2017-06-13:
Upper Wishbone のボルトがバンプストローク（Z+50mm）の時に Rr Fender に干渉するから
ポイントを少しずらして更に Better Geometry をと思って検討したけど，目論見が外れる... orz
[image]

y-yosuke, 2017-06-13:
やはり Rr Fender 側でなんとかするしかないか...

y-yosuke, 2017-06-15:
Spherical / Rodend Angle

- Lwr Wishbone Inner : 12.0613 deg

y-yosuke, 2017-06-21:
Dimple on Rr Fender for NAS Bolt - Rr Upper Wishbone / Rr Upright
[image]

y-yosuke, 2021-09-01:
Insights into bump-stop tuning
https://www.vehicledynamicsinternational.com/features/insights-into-bump-stop-tuning.html

y-yosuke, 2022-08-18:
- ホイールストロークhttps://www.weblio.jp/content/%E3%83%9B%E3%82%A4%E3%83%BC%E3%83%AB%E3%82%B9%E3%83%88%E3%83%AD%E3%83%BC%E3%82%AF
  - "一般の乗用車では通常150～200mmくらいである。"

現在前後とも ±50mm でストローク 100mm で設計しているが... 短いのか？

y-yosuke, 2022-09-18:
スフェリカルは結構大きいものを使っているように見える．

- Revealing the engineering secrets behind the Aston Martin Valkyrie - Words: Jonathan Bell
  - https://magazine.astonmartin.com/design/revealing-engineering-secrets-behind-aston-martin-valkyrie

y-yosuke, 2023-05-02:
- 世界初 常識外の挙動を実現する力学的非対称ゲル 理研が開発【橋本幸治の理系通信】- テレ東BIZ
  - https://txbiz.tv-tokyo.co.jp/txn/kaisetsu/post_273332