y-yosuke/formula-ppoino #38: CFD Radiator / Duct

y-yosuke, 2018-03-09:
Radiator CFD Model として Porous Media を設定したいが
そのための cellZone 有りのメッシュ作成に失敗するので単純な多孔モデルとしてみる．

- 20mm x 20mm Square Holes / 30mm Pitch
- Radiator - Control
  - Radiator : Area = 120523.294 (+/- 0.00023) square millimeters
  - Not Hole : Area = 70123.2943 (+/- 0.00023) square millimeters
  - Open about 41.8% : (120523.294-70123.2943)/120523.294 = 0.41817642073
- Radiatro C
  -

y-yosuke, 2018-03-09:
Mesh around the radiator in "Mesh - fp-024a - Control / Radiator Holes"
[image]

y-yosuke, 2018-03-10:
- Radiator Holes
  - Radiator Section Flow : 0.172154

y-yosuke, 2018-03-11:
Mesh around the radiator in  "Mesh - fp-024a - Control / Radiator Holes - 10mm x 10mm"
[image]

y-yosuke, 2018-03-12:
- Sidepod Control - C / Radiator Flow Comparison
  - Control Sidepod / Radiator Holes 10mm x 10mm - 28x18 : 0.13570  [m^3/s]
  - Sidepod C / Radiator Holes 10mm x 10mm - 28x20 : 0.15506 [m^3/s]

emijah, 2018-03-13:
どうしても大きめな穴になってしまうということですね？三角はダメですかね？

y-yosuke, 2018-03-13:
メッシュを細かくしたら穴は小さくできると思いますが 
20mm x 20mm と 10mm x 10mm では車両全体の空力性能値にあまり違いが出なかったので
10mm x 10mm ぐらいでとりあえずデバイス比較をしようかと考えています．

四角いのは SimScale の CFD メッシュが Hex Mesh なので
少ないメッシュで小さな穴が作成できるかと思いそうしています．

- Radiator Holes 10mm x 10mm
  - Control : Radiator Holes 20mm x 20mm

Body and Wheels
	
Coef.	|	Control	|	Radiator Holes 10mm x 10mm	|	Difference	|	Difference %	|	Remarks
---	|	---	|	---	|	---	|	---	|	---
Cm	|	0.0353	|	0.0356	|	0.0003	|	0.9	|	
Cd	|	0.5637	|	0.5638	|	0.0000	|	0.0	|	
Cl	|	-0.1810	|	-0.1806	|	0.0004	|	-0.2	|	
Clf	|	-0.0552	|	-0.0547	|	0.0005	|	-0.9	|	
Clr	|	-0.1258	|	-0.1259	|	-0.0002	|	0.1	|	
CoP	|	0.6950	|	0.6973	|	0.0022	|	0.2	|	
L/D	|	-0.3210	|	-0.3203	|	0.0007	|	-0.2	|	

Body Only

Coef.	|	Control	|	Radiator Holes 10mm x 10mm	|	Difference	|	Difference %	|	Remarks
---	|	---	|	---	|	---	|	---	|	---
Cm	|	0.0101	|	0.0104	|	0.0003	|	3.3	|	
Cd	|	0.4548	|	0.4548	|	0.0000	|	0.0	|	
Cl	|	-0.2190	|	-0.2188	|	0.0002	|	-0.1	|	
Clf	|	-0.0994	|	-0.0990	|	0.0004	|	-0.4	|	
Clr	|	-0.1196	|	-0.1198	|	-0.0002	|	0.2	|	
CoP	|	0.5461	|	0.5476	|	0.0016	|	0.2	|	
L/D	|	-0.4815	|	-0.4811	|	0.0005	|	-0.1	|	

Also see Google Drive file : 
https://docs.google.com/spreadsheets/d/1zDkpTxiz5o6DTmW6s7C5rSQ0SUoApb7MKp-XtCqKJaE/edit#gid=0

y-yosuke, 2018-03-14:
SimScale の Forum を見ると cellZone の作成失敗の件が多いからもしかして中の問題なのか？？？
https://www.simscale.com/forum/

y-yosuke, 2018-03-21:
（計算結果が貯まっていくばかりで... あぁ... まとめないと... 忘れる前にまとめないと...）

emijah, 2018-03-21:
:D ぜひ

y-yosuke, 2018-03-21:
SimScale のリンクをまだ貼っていなかった．
fp-024a - Sidepod / Radiator
https://www.simscale.com/workbench/?pid=4135868279315342050#tab_0-0

y-yosuke, 2018-03-23:
## fp024a - Sidepod / Radiator Study

This study is based on simulations as in below.

- Fp-023d - Rear Bodywork Study
  - Forum : https://www.simscale.com/forum/t/fp-023d-rear-bodywork-study/76182
  - Project : https://www.simscale.com/workbench/?pid=2563999609424123969#tab_0-0
- Fp-

### Contents

y-yosuke, 2018-03-23:
（レポートを書き始めましたので頻繁に過去の投稿を編集します）

y-yosuke, 2018-03-23:
もとい，Wiki ページでレポートを編集します．
https://github.com/y-yosuke/formula-ppoino/wiki/fp024a---Sidepod---Radiator-Study

y-yosuke, 2018-03-24:
Images of Simulation "Model with Updates"
[image]
[image]

y-yosuke, 2018-03-25:
Radiator Models
[image]
[image]
[image]

y-yosuke, 2018-03-25:
Sidepod C & Radiator Duct C Difference Images
[image]
[image]

y-yosuke, 2018-03-26:
Sidepod C - Louvers
[image]
[image]
[image]
[image]
[image]

y-yosuke, 2018-03-27:
Radiator Flow - U Magnitude
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

emijah, 2018-03-27:
> Radiator Flow - U Magnitude
> [image: fp024a_cfd_radiator_u-mag_sidepod-b_no-radiator]
> <https://user-images.githubusercontent.com/3119480/37950071-40f5ea14-31d2-11e8-9661-3a236b8d35ad.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-b_radiator-20mmx20mm]
> <https://user-images.githubusercontent.com/3119480/37950072-411f42ec-31d2-11e8-8667-f029f5ebf5c4.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-b_radiator-holes-10mmx10mm]
> <https://user-images.githubusercontent.com/3119480/37950073-41459bb8-31d2-11e8-9c16-507a239423fc.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_louver-lwr_bwd_top-half]
> <https://user-images.githubusercontent.com/3119480/37950074-416bfc72-31d2-11e8-9d0b-b8b90bc8cd9c.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_louver-lwr_bwd]
> <https://user-images.githubusercontent.com/3119480/37950075-4191e9a0-31d2-11e8-91c1-a17eda1f920c.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_louver-lwr_top-half]
> <https://user-images.githubusercontent.com/3119480/37950076-41ba138a-31d2-11e8-9775-b1bbc94fe108.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_louver-lwr]
> <https://user-images.githubusercontent.com/3119480/37950077-41e47454-31d2-11e8-89e3-149e0a2667eb.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_louver-upr]
> <https://user-images.githubusercontent.com/3119480/37950078-420c2b8e-31d2-11e8-9970-a10c0d9410e1.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_radiator-duct-d_turning-fillet]
> <https://user-images.githubusercontent.com/3119480/37950079-42341964-31d2-11e8-9bee-b883fee033b2.png>
> [image:
> fp024a_cfd_radiator_u-mag_sidepod-c_radiator-duct-d_turning-vane-h-b]
> <https://user-images.githubusercontent.com/3119480/37950080-4286bed0-31d2-11e8-82df-9c5578a2725d.png>
> [... cut ...]
> [image: fp024a_cfd_radiator_u-mag_sidepod-c_radiator-duct-d]
> <https://user-images.githubusercontent.com/3119480/37950084-433e2aca-31d2-11e8-819f-ab544c46e62b.png>
> [image: fp024a_cfd_radiator_u-mag_sidepod-c]
> <https://user-images.githubusercontent.com/3119480/37950085-4363c4c4-31d2-11e8-9f3d-98495d83b582.png>

ラジエータのリア側のダクト部分がカクカクになっているのですが、ここは丸くしてラジエータにもう少し半時計回りに開店させることはできないですかね？

emijah, 2018-03-27:
> fp024a_cfd_radiator_u-mag_sidepod-c_radiator-duct-d_turning-fillet]

NACAダクトのみでラジエータに空気を送るのはさすがに流量が少なすぎてダメですかね？

2018-03-27 15:44 GMT+09:00 Hajime Saito <emijah.s@gmail.com>:

y-yosuke, 2018-03-27:
@emijah フィレット形状のことでしょうか？
Sidepod C - Radiator Duct D / Turning Fillet

また，ラジエタのレイアウトは配管のことも考慮してモノコックに近すぎないようにしています．

y-yosuke, 2018-03-27:
NACAダクト風は 1994年の Ferrari 412T がありますが，
この形状は冷却に問題があり第5戦からコンベンショナルな形状に変わっています．

Ferrari 412T 初期バージョン
http://www.motorsportretro.com/2014/06/photo-of-the-day-1994-ferrari-412t/

https://ja.wikipedia.org/wiki/%E3%83%95%E3%82%A7%E3%83%A9%E3%83%BC%E3%83%AA%E3%83%BB412T1

y-yosuke, 2018-03-27:
今回の fp024a の CFD から流入空気は十分で
アウトレット側の空け方にラジエタ流量が大きく影響されているので
ラジエタダクト入り口の下側や外側にアタッチメントを設けて 
CLや  L/D の向上につながる可能性はあります．

こういうの（Radiator Duct の B にはデータを作成していたのですが Duct C と D には未作成）

[image]

一応，3月中にレポートとして SimScale に投稿したいので今はレポートを書くのを優先しています．

y-yosuke, 2018-03-27:
ラジエタのレイアウトはラジエタ部品により再考の余地はありますが「少しだけ」ですね．

- 上部をどれくらい傾けるか
- 前後位置
- 下後部はモノコックに対して十分近づいているので内側には行かない

お金をかけて非長方形のひし形や曲線外形のラジエタを作るのなら話は別ですが．

y-yosuke, 2018-03-27:
Image of Radiator Duct D / Vertical Turning Fillet / Vertical Turning Vane x1 
[image]

emijah, 2018-03-27:
インボード側にあるダクトの最後端ですよね？空気を曲げるにはRをつけた方が良さそうな感じがするのですが、間違いですかね？

emijah, 2018-03-27:
あ、しっかりとRがついていますね。失礼しました。

y-yosuke, 2018-03-27:
Fillet あり/なし も比較しましたが「あり」の方が 1% ほど良いですが， 
アウトレットのデバイスの効果にくらべたら誤差程度ですね．

型割り的に Fillet は別部品になると思うので 
Fillet なしでも 1% 程度しか悪くならないのであれば無くても良いかなと言う程度と考えています．
競技用ならつけるとは思います．という部品かな...

y-yosuke, 2018-03-27:
ざっというと

- ダクト内の改善効果 : 0〜3%
- アウトレットの改善効果 : 20〜50%

なので，現状の設計では流入側はサチっていて
空力的悪影響の少ない流出側の確保がラジエタ流量の向上する場合には必要と考えています．

（2000年代前半に多い「チムニー」はあまり好きではないので設計していない）

y-yosuke, 2018-03-27:
間違えた．逆でした．
"Fillet あり/なし も比較しましたが「あり」の方が 0.3% 流量が少ない"

なので Fillet は無しですね．

y-yosuke, 2018-03-27:
（ラジエタのモデリングが適切か？という問題もあるんだけど... そこらは風洞や実車との比較で確認して... てなってしまうので，まぁ「アタリ」をつける意味合いの CFD 解析ということで）

emijah, 2018-03-27:
少ないんだ！なんと。

emijah, 2018-03-27:
確かにチムニーはいまいちな感じですね。カウルの内側にNACAダクトは吸い出しに効くのだろうか？

y-yosuke, 2018-03-27:
@emijah そうなんですよ．誤差の誤差程度に少なかったんですよ．
なのでラジエタモデルにも依存しそうなぐらいの差ですし，まぁ今のとこと Fillet 無しで良いかなと．
検討は風洞や実車との相関が得られてからでも遅くないと思います．

y-yosuke, 2018-03-27:
吸い出しNACA風ダクトは 90年代初頭 に流行ったように思います．
イメージしているのとは逆方向かもしれませんが．
https://ja.wikipedia.org/wiki/%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB:Ayrton_Senna_1992_Monaco.jpg
http://motor.geocities.jp/formura1_model/F92ATkit.html

y-yosuke, 2018-03-27:
NACA Duct はある程度流体の運動エネルギーがあるところでないと効果が薄いイメージです．

y-yosuke, 2018-03-27:
NACA Duct が有効に働く条件が NACA の NACA Duct のレポートに書いてあったと思います．

y-yosuke, 2018-03-27:
Radiator Duct C も D も大して変わらないから Intake Adjustor Panel は C ので計算する．

y-yosuke, 2018-03-27:
（追加計算はネタ的にそのまますぎるので中止）

emijah, 2018-03-28:
NACA
ductを逆向きに使用する例はかなりありますが、なんとなく裏返しな感じがしてしまうんですよね。吸い出しにサメのエラのようなルーバーを使うケースが多いですが、なんとなく見た目だけな気がしてしまうんですよね。exhaust
ductについてちょっと調査してみます。

y-yosuke, 2018-03-28:
Pictures of Radiator Duct C and D
[image]
[image]

y-yosuke, 2018-03-28:
Sidepod C - Pressure / Left View
[image]

y-yosuke, 2018-03-28:
Sidepod C - Louvers / Pressure Images
[image]
[image]

y-yosuke, 2018-03-28:
Sidepod C - Louvers / U Magnitude at Z=220mm Capture Image
[image]

emijah, 2018-03-28:
さらにリップをつけると良さそうですね。

y-yosuke, 2018-03-29:
単なる穴やチムニーだと静止時に雨が入ってくるので
ルーバーでかつリアタイアフェアリングの下に来るのが良いかな．

位置は浸水のことも考えると高い位置のほうが良いし冷却能力が十分であれば
Louver Lwr / Top Half が効率的で良いと考えている．

y-yosuke, 2018-03-29:
（浸水だったら Rr Lwr Wishbone の通る穴が低いではないか．というのはある）

y-yosuke, 2018-03-29:
Turning Fillet
[image]

Turning Vane - Vertical
[image]

Turning Vane - Vertical 2 pcs
[image]

Turning Vane - Horizontal
[image]

Turning Vane - Horizontal B
[image]

y-yosuke, 2018-03-29:
とりあえずレポートを一通り書いた．
https://github.com/y-yosuke/formula-ppoino/wiki/%5Bfp024a%5D-Sidepod-Radiator-Study

何時間かしてから読み返して修正して SimScale に投稿する．

y-yosuke, 2018-03-29:
markdown の Table に SimScale が対応していて嬉しい．