y-yosuke/formula-ppoino #41: [CFD] fp024c - Side Mirrors

y-yosuke, 2018-08-21:


y-yosuke, 2018-08-21:
Now editing Wiki
- https://github.com/y-yosuke/formula-ppoino/wiki/%5Bfp024c%5D-Side-Mirror-Comparison

y-yosuke, 2018-08-21:
[image]
[image]
[image]
[image]
[image]

y-yosuke, 2018-09-20:
[image]
[image]
[image]

y-yosuke, 2018-09-21:
[image]
[image]
[image]
[image]

y-yosuke, 2018-09-21:
[image]

emijah, 2018-09-21:
面白い

y-yosuke, 2018-09-21:
Editing now... 
https://github.com/y-yosuke/formula-ppoino/wiki/%5Bfp024c%5D-Side-Mirror-Comparison-(Passive-Flow-Jet-Control)

y-yosuke, 2018-09-22:
Side Mirrorless 22[m/s] Contour Surface and Side Mirror B
[image]

y-yosuke, 2018-09-23:
一通り書いた．明日また見直して SimScale Forum に投稿予定．
https://github.com/y-yosuke/formula-ppoino/wiki/%5Bfp024c%5D-Side-Mirror-Comparison-(Passive-Flow-Jet-Control)

y-yosuke, 2019-08-17:
Passive flow jet control side mirrors (and bodyworks?) on Bugatti.
https://www.youtube.com/watch?v=0BsJgvXVfWU

y-yosuke, 2022-08-27:
- Red Bull Racing RB18 mirror detail
  - https://cdn-1.motorsport.com/images/mgl/27vPlzX0/s1200/red-bull-racing-rb18-mirror-de-1.webp